polarfire-soc/hart-software-services #2: build failue in mpfs_hal (gcc 10.1)

leiflindholm, 2020-08-29:
When attempting to build BOARD=icicle-kit-es with the current version of
https://github.com/riscv/riscv-gnu-toolchain (gcc 10.1.0), it bombs out in 
baremetal/polarfire-soc-bare-metal-library/src/platform/mpfs_hal/system_startup.c with:

`baremetal/polarfire-soc-bare-metal-library/src/platform/mpfs_hal/system_startup.c:115:25: error: array subscript -64 is outside array bounds of 'long unsigned int[1]' [-Werror=array-bounds]
  115 |                     hls = (HLS_DATA*)((uint8_t *)&__stack_bottom_h1$
      |                     ~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  116 |                             + (((uint8_t *)&__stack_top_h1$ -
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  117 |                                     (uint8_t *)&__stack_bottom_h1$) * hard_idx)
      |                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  118 |                                 - (uint8_t *)(HLS_DEBUG_AREA_SIZE));
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from baremetal/polarfire-soc-bare-metal-library/src/platform/mpfs_hal/mss_hal.h:70,
                 from baremetal/polarfire-soc-bare-metal-library/src/platform/mpfs_hal/system_startup.c:32:
baremetal/polarfire-soc-bare-metal-library/src/platform/mpfs_hal/system_startup.h:41:22: note: while referencing '__stack_bottom_h1$'
   41 | extern unsigned long __stack_bottom_h1$;
      |                      ^~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors`

lbmeng, 2020-08-30:
Yes this seems to be a known issue with the codes working with GCC 10.x.

Currently I am using GCC 9.2.0 to build the HSS.

griffini, 2020-08-31:
Thanks guys for reporting this error.

For right now, I would suggest that the best way to build the HSS is to use the compilers that come with Microchip's SoftConsole 12.4 or later, as this is the most tested platform, and it will also be the preferred mechanism for reprogramming the eNVM on the Icicle Kit.

Having said that, we are currently looking at fixing the particular error above, and will push the fix once we have it.

leiflindholm, 2020-08-31:
Thanks.

For reference: using https://github.com/riscv/riscv-gnu-toolchain commit 7f1f4ab5b0e0 (last before switch to gcc 10.1), I produce a working gcc 9.2.0 with which I can successfully build hss.bin for BOARD=icicle-kit-es-sd and run under QEMU with @lbmeng's platform support patches (after commenting out HSS_DDRInit from the board init, due to DDR controller not being emulated).

griffini, 2020-09-01:
> Thanks.
>
> For reference: using https://github.com/riscv/riscv-gnu-toolchain commit 7f1f4ab5b0e0 (last before switch to gcc 10.1), I produce a working gcc 9.2.0 with which I can successfully build [...]

Sounds good!

Just FYI, in case it is of interest, perhaps you know that RENODE is a UI and set of extensions around QEMU that make it easier to simulate hardware and peripherals. RENODE is bundled with the SoftConsole development tools from Microchip.

However, RENODE1.10.1 includes enough support for the DDR parts of PolarFire SoC to allow the HSS to run *without* needing to hack out HSS_DDRInit().  You can find RENODE 1.10.1 at https://github.com/renode/renode/releases

To run the demo run:
“include @scripts/single-node/icicle kit.resc”
... followed by
“start”

And for completeness sake, the fake registers for DDR: https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/Miscellaneous/MPFS_DDRMock.cs

lbmeng, 2020-09-01:
AFAIK, Renode 1.10.1 does not support running the latest HSS out of the box.

griffini, 2020-09-01:
Hi,

Yes, it certainly should — Renode release 1.10.1 can be used for running Linux on PolarFire SoC. It will be included in the upcoming SoftConsole release.  A standalone installer can be found at: https://github.com/renode/renode/releases/tag/v1.10.1

A script is included to run the PolarFire SoC Linux demo... type the following into Renode:

```
include @scripts/single-node/icicle-kit.resc
start
```

This will download a pre-canned demo with prebuilt binaries (but you can modify the .resc file and build the assets yourself). This includes starting the HSS, and use it to load an image from simulated eMMC. The HSS will output to UART0, openSBI, U-Boot and the Linux console will output to UART1.

To help speed up boot you can reduce the log level for the system by globally setting the log level at the Renode prompt:
```
logLevel 3
```

lbmeng, 2020-09-14:
> Yes, it certainly should — Renode release 1.10.1 can be used for running Linux on PolarFire SoC. It will be included in the upcoming SoftConsole release. A standalone installer can be found at: https://github.com/renode/renode/releases/tag/v1.10.1

The image included in Renode 1.10.1 was probably built from an earlier version of HSS, as I tested one image built from the latest HSS code base but it did not boot in Renode 1.10.1. :(

griffini, 2020-09-14:
> The image included in Renode 1.10.1 was probably built from an earlier version of HSS, as I tested one image built from the latest HSS code base but it did not boot in Renode 1.10.1. :(

Ah, thanks for spotting this.  Can I suggest you try the following - in the RENODE console, type:

````
e51 LogFunctionNames
````

This should show you where the HSS is getting stuck. My guess is that it is in checking a register bit somewhere that is not currently modelled in RENODE.

griffini, 2020-09-18:
Just to note that a fix for the original issue (gcc-10 compile problem) has been merged in...

( See https://github.com/polarfire-soc/hart-software-services/commit/43e3fffe29bc28318a7f3d5aa9d9507918c92444.)

Cyril-Jean, 2020-09-18:
Issue fixed in tag 0.99.11.

closed 2020-09-18 by Cyril-Jean

lbmeng, 2020-09-23:
> Just to note that a fix for the original issue (gcc-10 compile problem) has been merged in...

Yes, the build failure with GCC 10 has been fixed, but the image does not boot. Please have a look. Thanks!