polarfire-soc/hart-software-services #3: HSS built by GCC 10.1.0 does not boot

lbmeng, 2020-09-23:
https://github.com/polarfire-soc/hart-software-services/issues/2 addressed the build failure of HSS when built with GCC 10.1.0

However the HSS image built by GCC 10.1.0 does not boot. Changing compiler back to GCC 9.2.0, the image boots again.

griffini, 2020-09-23:
Thank you for the bug report, we'll take a look.

When you said it does not boot, do you mean on the Icicle kit or on RENODE?

lbmeng, 2020-09-23:
On QEMU. RENODE does not boot GCC 9.2.0 imag too but that's a RENODE emualtion issue.

griffini, 2020-09-23:
Hmm... I just built a GCC 10.1.0 cross-compiler from source, and compiled the HSS with it. The resulting image appears to work fine on an Icicle kit board. 

[image]

Likewise with RENODE (if I work around a spin on an unimplemented MMC status bit)...

Are you perhaps building for target other than BOARD=icicle-kit-es ?

lbmeng, 2020-09-25:
> Hmm... I just built a GCC 10.1.0 cross-compiler from source, and compiled the HSS with it. The resulting image appears to work fine on an Icicle kit board.

Thanks for the confirmation on the real board.

> I would expect things not to work on QEMU as it won't have any of the hardware simulated -- unless, are you perhaps building for target other than BOARD=icicle-kit-es?

The latest QEMU has the Microchip Icicle Kit support. I am building a HSS with BOARD=icicle-kit-es. The weird thing is that using GCC 9.2.0 it boots on QEMU but with GCC 10.1.0 it does not. I am now debugging the GCC 10.1.0 image.

griffini, 2020-09-25:
Ah, I see you added support for it in qemu (in hw/riscv/microchip_pfsoc.c) ...

This is really cool!

lbmeng, 2020-09-25:
> Ah, I see you added support for it in qemu (in hw/riscv/microchip_pfsoc.c) ...
> 
> This is really cool!

:)

I figured out what issue is. Please see https://github.com/polarfire-soc/hart-software-services/pull/6
I am curious why the image you built with GCC 10 can boot on the real board.

griffini, 2020-09-25:
Yes, thank you for the pull request. I think it might be better to fix it in the linker script, as it appears .srodata is placed wrong.

Why it worked for me on the board is because I think I must have built it to run from L2LIM (scratchpad memory), where there are no copies to the correct locations... whereas you were running the eNVM build (default linker script), which boots from eNVM flash on a real board.   That way, I wasn't seeing the placement as an issue.  Great catch, thank you!

lbmeng, 2020-09-28:
Merged by https://github.com/polarfire-soc/hart-software-services/commit/d8177c06ef83ee57fa84eef017492007ffdae6aa

closed 2020-09-28 by lbmeng

sarah18-z, 2024-07-09:
Hello @lbmeng, 
Please, I would like to know how you used QEMU to emulate the Icicle Kit board. I tried to do it, but it indicates that the board is not supported by the emulator, and I can't find anything in the documentation that can help me.
[image]