polarfire-soc/hart-software-services #8: build fails on debian buster system

bjdooks-ct, 2020-10-11:
using cross-building from a debian-buster x86_64 the build fails on crt.S

$  make CROSS_COMPILE=riscv64-linux-gnu- 
crt.S: Assembler messages:
crt.S:23: Error: unknown pseudo-op: `.attribute'
crt.S:24: Error: unknown pseudo-op: `.attribute'

$ riscv64-linux-gnu-gcc -v
gcc version 8.3.0 (Debian 8.3.0-2)

griffini, 2020-10-12:
I think the problem is that your tools are slightly too old. In particular, the assember, which isn't recognising .attribute.

For reference, the HSS will always compile with the tools provided with Microchip SoftConsole - https://www.microsemi.com/product-directory/design-tools/4879-softconsole#downloads

At the current time, these are:

```
$ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=/home/igriffin/Microchip/scLinux-6.4.0.407-20200826-105139/riscv-unknown-elf-gcc/bin/riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/igriffin/Microchip/scLinux-6.4.0.407-20200826-105139/riscv-unknown-elf-gcc/bin/../libexec/gcc/riscv64-unknown-elf/8.3.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /Host/home/user/Work/riscv64-unknown-elf-gcc-8.3.0-1.1/sources/riscv-gcc-8.3.0-1.1/configure --prefix=/Host/home/user/Work/riscv64-unknown-elf-gcc-8.3.0-1.1/linux-x64/install/riscv64-unknown-elf-gcc --infodir=/Host/home/user/Work/riscv64-unknown-elf-gcc-8.3.0-1.1/linux-x64/install/riscv64-unknown-elf-gcc/share/doc/info --mandir=/Host/home/user/Work/riscv64-unknown-elf-gcc-8.3.0-1.1/linux-x64/install/riscv64-unknown-elf-gcc/share/doc/man --htmldir=/Host/home/user/Work/riscv64-unknown-elf-gcc-8.3.0-1.1/linux-x64/install/riscv64-unknown-elf-gcc/share/doc/html --pdfdir=/Host/home/user/Work/riscv64-unknown-elf-gcc-8.3.0-1.1/linux-x64/install/riscv64-unknown-elf-gcc/share/doc/pdf --build=x86_64-unknown-linux-gnu --host=x86_64-unknown-linux-gnu --target=riscv64-unknown-elf --with-pkgversion='xPack GNU RISC-V Embedded GCC (Microsemi SoftConsole build)\x2C 64-bit' --with-bugurl=https://github.com/sifive/freedom-tools/issues --enable-languages=c,c++ --disable-mingw-wildcard --enable-plugins --enable-lto --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-threads --disable-tls --enable-checking=yes --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-riscv64-unknown-elf --with-sysroot=/Host/home/user/Work/riscv64-unknown-elf-gcc-8.3.0-1.1/linux-x64/install/riscv64-unknown-elf-gcc/riscv64-unknown-elf --with-native-system-header-dir=/include --with-abi=lp64d --with-arch=rv64imafdc --disable-rpath --disable-build-format-warnings --with-system-zlib
Thread model: single
gcc version 8.3.0 (xPack GNU RISC-V Embedded GCC (Microsemi SoftConsole build), 64-bit) 
```

But interestingly, the assembler is 2.32:

```
$ riscv64-unknown-elf-as --version
GNU assembler (xPack GNU RISC-V Embedded GCC (Microsemi SoftConsole build), 64-bit) 2.32
Copyright (C) 2019 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `riscv64-unknown-elf'.
```

The HSS should also compile with the latest RISC-V GCC tools from https://gcc.gnu.org/git/

bjdooks-ct, 2020-10-12:
Thanks, will try and build the microchip ones tomorrow.
I also tried with gcc-10 on debian/testing, but that builds and fails to run (possibly there's some other issue).

I've got a couple of followup questions:
- does the hart-software-services print anything at init time? if not, is there an option to
- if it does, which qemu serial port should I be watching
- it would be great if the above could be in the documentation about running with qemu

bjdooks-ct, 2020-10-12:
I've tried with SoftCosole-v6.4 and that builds and seems to print stuff even if it doesn't boot my sd card image.

griffini, 2020-10-13:
Okay, great.

Does it print an error message for your SD card image?

griffini, 2020-10-13:
> Thanks, will try and build the microchip ones tomorrow.
> I also tried with gcc-10 on debian/testing, but that builds and fails to run (possibly there's some other issue).

There was an issue previously (w.r.t. where the srodata section was being placed) but that has been resolved. 

> 
> I've got a couple of followup questions:
> 
> * does the hart-software-services print anything at init time? if not, is there an option to
> * if it does, which qemu serial port should I be watching
> * it would be great if the above could be in the documentation about running with qemu

On PolarFire SoC there are 5 UARTs, which essentially allows each Hart to "own" one (each Hart has one UART it can map  interrupts local to its CLINT, and all UART interrupts can alternatively be mapped globally via the PLIC).

See https://wiki.qemu.org/Documentation/Platforms/RISCV#Microchip_PolarFire_SoC_Icicle_Kit which mentions that, when using their command line string, that "HSS output is on the first serial port (stdio) and U-Boot/Linux outputs on the 2nd serial port. OpenSBI outputs on a random serial port due to the lottery mechanism used during the multi-core boot."

bjdooks-ct, 2020-10-13:
I did:

 zcat ./yocto.ice/build/tmp-glibc/deploy/images/icicle-kit-es-sd/mpfs-dev-cli-icicle-kit-es-sd-20201012103345.rootfs.wic.gz > sd.img
qemu-img resize sd.img 64G
./qemu/bin/qemu-system-riscv64 -M microchip-icicle-kit -smp 5  -bios hart-software-services/Default/hss.bin -sd sd.img -display none -serial stdio -chardev socket,id=serial1,path=serial1.sock,server,wait -serial chardev:serial1 -m 2G

But it ends at:
[0.42031] HSS_MemTestDDRFast(): DDR size is 1 GiB

I have also got running:
socat unix:serial1.sock -
but nothing comes out of there.

bjdooks-ct, 2020-10-13:
ok, so looks like it ended up configured for emmc boot instead of sd, now i've configured sd I get the followingL

U-Boot 2020.01 (Oct 12 2020 - 10:49:15 +0000)

DRAM:  1 GiB
exception code: 7 , Store/AMO access fault , epc bff941c0 , ra bff9f060
### ERROR ### Please RESET the board ###

griffini, 2020-10-16:
Very Interesting. The Store/AMO access fault is a result of a failing PMP check in QEMU.

So it seems this is nothing to do with the HSS, but rather an interaction between QEMU and U-Boot.  The latest PolarFire-SoC U-Boot patches include some code changes that are accessing a memory region that QEMU hasn't yet modelled.

I've posted a suggested patch to the QEMU mailing list, but if you want to try it yourself, the patch is :

```
 hw/riscv/microchip_pfsoc.c         | 6 ++++++
 include/hw/riscv/microchip_pfsoc.h | 1 +
 2 files changed, 7 insertions(+)

diff --git a/hw/riscv/microchip_pfsoc.c b/hw/riscv/microchip_pfsoc.c
index 4627179cd3..20e1496e3e 100644
--- a/hw/riscv/microchip_pfsoc.c
+++ b/hw/riscv/microchip_pfsoc.c
@@ -97,6 +97,7 @@ static const struct MemmapEntry {
     [MICROCHIP_PFSOC_GPIO2] =           { 0x20122000,     0x1000 },
     [MICROCHIP_PFSOC_ENVM_CFG] =        { 0x20200000,     0x1000 },
     [MICROCHIP_PFSOC_ENVM_DATA] =       { 0x20220000,    0x20000 },
+    [MICROCHIP_PFSOC_IOSCB_CTRL] =      { 0x37020000,     0x1000 },
     [MICROCHIP_PFSOC_IOSCB_CFG] =       { 0x37080000,     0x1000 },
     [MICROCHIP_PFSOC_DRAM] =            { 0x80000000,        0x0 },
 };
@@ -341,6 +342,11 @@ static void microchip_pfsoc_soc_realize(DeviceState *dev, Error **errp)
     create_unimplemented_device("microchip.pfsoc.ioscb.cfg",
         memmap[MICROCHIP_PFSOC_IOSCB_CFG].base,
         memmap[MICROCHIP_PFSOC_IOSCB_CFG].size);
+
+    /* IOSCBCTRL */
+    create_unimplemented_device("microchip.pfsoc.ioscb.ctrl",
+        memmap[MICROCHIP_PFSOC_IOSCB_CTRL].base,
+        memmap[MICROCHIP_PFSOC_IOSCB_CTRL].size);
 }

 static void microchip_pfsoc_soc_class_init(ObjectClass *oc, void *data)
diff --git a/include/hw/riscv/microchip_pfsoc.h b/include/hw/riscv/microchip_pfsoc.h
index 8bfc7e1a85..3f1874b162 100644
--- a/include/hw/riscv/microchip_pfsoc.h
+++ b/include/hw/riscv/microchip_pfsoc.h
@@ -95,6 +95,7 @@ enum {
     MICROCHIP_PFSOC_ENVM_CFG,
     MICROCHIP_PFSOC_ENVM_DATA,
     MICROCHIP_PFSOC_IOSCB_CFG,
+    MICROCHIP_PFSOC_IOSCB_CTRL,
     MICROCHIP_PFSOC_DRAM,
 };

```

Cyril-Jean, 2020-10-16:
OK, this makes sense to me. This is likely because U-Boot is now retrieving the device's serial number from the system controller to create a unique MAC address. This is the first time we would have accessed the system controller from the Linux world.
The U-boot patches are in-flight so this QEMU patch will be required to test the submitted U-Boot patches.

bjdooks-ct, 2020-10-18:
Thanks. Resulting u-boot is now working:

```
U-Boot 2020.01 (Oct 12 2020 - 10:49:15 +0000)

DRAM:  1 GiB
error setting mac-address property
MMC:   sdhc@20008000: 0
In:    serial@20100000
Out:   serial@20100000
Err:   serial@20100000
Net:   
Warning: ethernet@20112000 using MAC address from ROM
eth0: ethernet@20112000
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr.uimg
1096 bytes read in 13 ms (82 KiB/s)
## Executing script at 88100000
3586448 bytes read in 762 ms (4.5 MiB/s)
## Copying 'fdt@microchip_icicle-kit-es-a000-microchip.dtb' subimage from FIT image at 88300000 ...
sha256+    Loading part 253 ... OK
## Loading kernel from FIT Image at 88300000 ...
   Using 'conf@microchip_icicle-kit-es-a000-microchip.dtb' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  gzip compressed
     Data Start:   0x88300100
     Data Size:    3574556 Bytes = 3.4 MiB
     Architecture: RISC-V
     OS:           Linux
     Load Address: 0x80200000
     Entry Point:  0x80200000
     Hash algo:    sha256
     Hash value:   6c219a83a24340be3e4b2bfcc57b373c0a4c19857d2c72167a5b9f129f891af2
   Verifying Hash Integrity ... sha256+ OK
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Uncompressing Kernel Image
   Using Device Tree in place at 0000000088000000, end 0000000088005678

```

This is now loading the kernel, which then fails with the following:

```
[    0.413705] printk: console [ttyS0] enabled
[    0.414327] printk: bootconsole [sbi0] disabled
[    0.416585] 20102000.serial: ttyS1 at MMIO 0x20102000 (irq = 13, base_baud = 37500000) is a 16550A
[    0.418204] 20104000.serial: ttyS2 at MMIO 0x20104000 (irq = 14, base_baud = 37500000) is a 16550A
[    0.440119] loop: module loaded
[    0.461769] Rounding down aligned max_sectors from 4294967295 to 4294967288
[    0.463372] db_root: cannot open: /etc/target
[    0.465342] libphy: Fixed MDIO Bus: probed
[    0.466465] macb 20112000.ethernet: GEM doesn't support hardware ptp.
[    0.467864] libphy: MACB_mii_bus: probed
[    0.470825] macb 20112000.ethernet eth0: Cadence GEM rev 0x0107010c at 0x20112000 irq 16 (56:34:12:00:fc:00)
[    0.471430] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.471678] ehci-platform: EHCI generic platform driver
[    0.472014] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    0.472237] ohci-platform: OHCI generic platform driver
[    0.472766] usbcore: registered new interface driver cdc_acm
[    0.472958] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[    0.473740] i2c /dev entries driver
[    0.474601] Oops - load access fault [#1]
[    0.474777] Modules linked in:
[    0.475040] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.6.16 #1
[    0.475433] epc: ffffffe0002f934e ra : ffffffe0002f9344 sp : ffffffe03968dbf0
[    0.475723]  gp : ffffffe00061bc80 tp : ffffffe039690000 t0 : ffffffe038bce840
[    0.476000]  t1 : 0000000000000000 t2 : 0000000000000000 s0 : ffffffe03968dc30
[    0.476267]  s1 : ffffffe038c19418 a0 : 0000000000001770 a1 : 0000000000000000
[    0.476556]  a2 : 0000000000000000 a3 : ffffffe039690000 a4 : 00000000000186a0
[    0.476857]  a5 : ffffffd00401b000 a6 : 0000000000000001 a7 : 0000000000000000
[    0.477144]  s2 : 0000000000000000 s3 : ffffffe039771810 s4 : ffffffe039771800
[    0.477414]  s5 : ffffffe039086f60 s6 : ffffffe00061d298 s7 : ffffffe00061d290
[    0.477732]  s8 : ffffffe000019460 s9 : 0000000000000008 s10: ffffffe00000020c
[    0.478070]  s11: 0000000000000000 t3 : ffffffe038bce840 t4 : ffffffe0005e71d8
[    0.478328]  t5 : ffffffe0005e71e0 t6 : ffffffe038bce838
[    0.478526] status: 0000000000000120 badaddr: ffffffd00401b000 cause: 0000000000000005
[    0.479251] ---[ end trace 51676042a9102299 ]---
[    0.479578] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    0.479934] SMP: stopping secondary CPUs
[    0.480386] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---

```

griffini, 2020-10-19:
Progress of a sort, I guess.  I think at this point the problem is clearly not in the HSS, but in QEMU.  

@lbmeng has said on qemu-devel that he is currently adding the DDR controller modeling support to PolarFire SoC which will address the missing SCBCTRL region.

Given that, okay if we close this issue on the HSS?

lbmeng, 2020-10-19:
Thanks @griffini for the information. It looks like this thread has mixed issues reported.

Let's track other issues independently, e.g.: via QEMU mailing list if it's a problem of running latest HSS.

bjdooks-ct, 2020-10-19:
ok, looks like I will try waiting for the real hardware to arrive.

closed 2020-10-19 by bjdooks-ct

griffini, 2020-10-19:
In the interim, you could also try RENODE, which is an alternative emulated based on QEMU.  

The RENODE release 1.10.1 can be used for running Linux on PolarFire SoC. It will be included in the upcoming SoftConsole release. A standalone installer can be found at: https://github.com/renode/renode/releases/tag/v1.10.1

It relies on building the eMMC variant in Yocto, and may need one small patch (see the #if 0 below) to the HSS to avoid spinning on an unmodelled bit in the eMMC driver itself currently, but it will successfully boot Linux to a working console prompt.

```
static cif_response_t check_device_status(cif_response_t rsp_status)
{
#if 0
    uint32_t srs9;

    do
    {
        srs9 = MMC->SRS09;
    }while ((srs9 & SRS9_DAT0_SIGNAL_LEVEL) == MMC_CLEAR);
#endif

    /*
    * Writing to the EXT CSD register takes significant time,
    * so function must not return until the 'READY FOR DATA'
    * bit is set in the Card Status Register.
    */

    while (DEVICE_BUSY == rsp_status)
    {
        rsp_status = cif_send_cmd(sdcard_RCA << RCA_SHIFT_BIT,
                                    MMC_CMD_13_SEND_STATUS,
                                    MSS_MMC_RESPONSE_R1);
    }

    return (rsp_status);
}
```