polarfire-soc/hart-software-services #9: [bin2chunks] handling privMode looks wrong...

yokoyama-flogics, 2020-10-23:
Hello,

Thank you for the great tools.

I'm now evaluating bin2chunks (commit 078ce16) to boot hart 1 and 2 (U54_1 and U54_2) as

```
$ ./tools/bin2chunks/bin2chunks 0x08020000 0x8020000 0 0 32768 payload.bin 1 3 some.bin 0x08020000 2 3 /dev/null 0
```

However, this causes an assertion error in ```validate_privmode_()```.

I have applied the following changes and it now looks working fine.

```diff
diff --git a/tools/bin2chunks/bin2chunks.c b/tools/bin2chunks/bin2chunks.c
index 32e6e2b..401b2cd 100755
--- a/tools/bin2chunks/bin2chunks.c
+++ b/tools/bin2chunks/bin2chunks.c
@@ -474,7 +474,7 @@ static void validate_owner_(int owner) {
 }
 
 static void validate_privmode_(int privmode) {
-    assert((privmode >= PRV_U) && (privmode <=PRV_S));
+    assert((privmode >= PRV_U) && (privmode <=PRV_M));
 }
 
 char imageNameBuf[1024] = "";
@@ -535,7 +535,7 @@ int main(int argc, char **argv)
         printf(" - hart owner is >>%d<<\n", ownerArray[i]);
 
         privMode[i] = strtol(argv[argIndex++], NULL, 10);
-        validate_privmode_(ownerArray[i]);
+        validate_privmode_(privMode[i]);
         printf(" - privMode >>%d<<\n", privMode[i]);
 
         pFileNameArray[i] = argv[argIndex++];
```

Could you please check the code?

Regards,
Atsushi Yokoyama

griffini, 2020-12-15:
Yes, this code will prevent booting into M mode. This was by design at the time.  This tool is now deprecated and has been replaced by the hss-payload-generator.

The updated hss-payload-generator tool will allow a payload to boot into M mode.

yokoyama-flogics, 2020-12-17:
Hi @griffini,

OK, I understood that.  I'm migrating to hss-payload-generator.

Please close this thread.

Regards,
Atsushi

closed 2020-12-17 by yokoyama-flogics