polarfire-soc/hart-software-services #11: DDR size?

kelas, 2021-01-22:
Apologies in advance for an overly naive question - I'm coming from a different background, and very new to FPGA kitchen. I have had the board for a while, but decided to give it a spin only now.   

So basically this:

1. The original [Crowdsupply advert](https://www.crowdsupply.com/microchip/polarfire-soc-icicle-kit) says the board comes with 2GB RAM:
```
...
2 GB LPDDR4 x 32
...
```

2. My board, however, boots up to half of that:

```
HSS_Boot_PMPSetupHandler(): Hart1 setup complete
HSS_OpenSBI_Setup(): MTVEC switching from 20220230 to 20220100
U-Boot 2020.01 (Aug 11 2020 - 21:55:51 +0000)
DRAM:  1 GiB
...
```

3. HSS code has the same opinion, e.g. memtest is hardcoded to 1GB:

https://github.com/polarfire-soc/hart-software-services/blob/078ce16d8cb3a89e4c3e93916621b816dbdaa274/misc/hss_memtest.c#L211

4. What I find most puzzling is the reading on the actual DDR chip, which is a **Micron D9WHZ**, which expands to:

```
MT53D512M32D2DS-053 WT:D	D9WHZ          16GB
```

Simply put, how much RAM can I address on this device? What am I overlooking?

griffini, 2021-01-28:
Hi @kelas, 

The device does have 2 GiB but the device trees in Yocto currently had this limited to 1GiB. 
We have tested it with the full 2 GiB and I expect a subsequent Yocto release will enable this.

The Micron part is a 16Gib parto, so 2 GiB (see https://www.micron.com/products/dram/lpdram/part-catalog/mt53d512m32d2ds-053-aut).

kelas, 2021-01-28:
Hello Ivan,

Thanks for clarification, in the meantime I figured out Yocto/HSS build issues on RHEL systems, and found a way to tweak the subject matter to my liking.

I'd appreciate if you could leave this issue open until there's the official release from Microchip which would unlock the  hardware to the spec. 

All in all, very interesting system to fiddle around with, even if a little over-engineered for my taste (tastes differ).

PCIe is particularly useful.

Keep up the good work!

k.

griffini, 2021-01-28:
Thanks,

I'm okay with leaving the issue open, but would it make more sense to raise it on https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp and/or https://github.com/polarfire-soc/polarfire-soc-buildroot-sdk, given that the device trees are there?

The HSS has no role in configuring how much DDR is available to the system.

griffini, 2021-01-28:
> Thanks for clarification, in the meantime I figured out Yocto/HSS build issues on RHEL systems, and found a way to tweak the subject matter to my liking.

BTW, I'd appreciate learning more about the HSS build issues on RHEL.

kelas, 2021-01-28:
> HSS build issues on RHEL.

in short, on CentOS 8 it blows up halfway into the build when `as` is being passed `--64` flag instead of `-m64` (it is the wrong `as`)

I ended up vandalising my env to make it use correct assembler quick and dirty, but I'm fairly certain this isn't "just me" issue.

Which distro you guys are using to run bitbake?

kelas, 2021-01-28:
> The HSS has no role in configuring how much DDR is available to the system.

Yes, but it makes assumptions about it, e.g. `memtest`.

> but would it make more sense to raise it on...

Presuming the other repo is responsibility of another team, yes — I'll open a dummy ticket there with a reference to this one, if that's ok.

griffini, 2021-01-29:
> > HSS build issues on RHEL.
> 
> On CentOS 8 it blows up halfway into the build when `as` is being passed `--64` (it is the wrong `as` which only understands `-m64`)
> 
> I ended up vandalising my env to make it use correct assembler, but I'm fairly certain this isn't "just me" issue.

Could you easily paste the error message? What was it assembling at the time?

> 
> Which distro you guys are using to run bitbake?

Most commonly I think Ubuntu LTS (18.04, 20.04).

griffini, 2021-09-23:
Hi @kelas, as there as been no update on this in a few months, I trust it is okay if I close this issue now. If you have continuing issues, please feel free to open a new ticket.

closed 2021-09-23 by griffini