polarfire-soc/hart-software-services #17: e.MMC/SDcard code overwrites unrelated fields in the IOMUXx_CR registers

ak-at-aries, 2021-03-15:
Code, switching the IO mode between e.MMC and SDcard interfaces (services/mmc/mmc_api.c), writes per-defined constants into the IOMUXx_CR (x=1, 2, 6) registers and overwrites settings for unrelated pads, making them non-functional.

Additionally, the documentation for the IOMUXx_CR registers seems to be incomplete, this issue is reported via Microsemi SoC support portal under case # 493642-2837732661

hughbreslin, 2021-03-15:
Hi @ak-at-aries you're correct this code is not 100% final, we figured out how to get this working before we could implement the changes in Libero to provide this feature - the values used here were a diff between what was generated for the eMMC and SD card configurations for the MPFS250 device configuration for the icicle kit and then hard coded. The plan would be for the mss configurator to have the option to add a "default" and "secondary" config options for peripherals so XML can be generated to support the switching feature which could the be accessed via a HAL API.

At present we don't have this support from the configurator so we went with the hard coded values to bring the switching feature forward while we wait for the additional functionality to be implemented. 

Can I check are you having an issue or just highlighting this?

ak-at-aries, 2021-03-16:
Hello @hughbreslin.

> Hi @ak-at-aries you're correct this code is not 100% final, we figured out how to get this working before we could implement the changes in Libero to provide this feature - the values used here were [...]

Thanks for the explanations and peek into the plans to support this functionality. It's understood
and makes [at least some] sense.

> At present we don't have this support from the configurator so we went with the hard coded values to bring the switching feature forward while we wait for the additional functionality to be implemented.

 There still might be better way to address it even now, e.g. masking
register content and changing only the bits that differs between e.MMC and SDcard modes... but since
this code is going to be changed anyway, it's OK as it is for the moment.

> Can I check are you having an issue or just highlighting this?

We have a working solution, no immediate issues here.

On our module we have external mux switching between the e.MMC (on the module)
and external SDcard socket, very similar to Icicle board, but controlled via MSS GPIO 12.
At the moment we have modified the code to work with our hardware and this
issue is "just for the records", to keep track of what we think may be improved.

You can see our current code at https://github.com/ak-at-aries/hart-software-services
and several PRs has already been submitted base on it, not this particular one, because
the code we have implemented is not generic and not any single bit better than your
-- it is the very same hard coded constants approach you choose, just with modified constants.

hughbreslin, 2021-03-16:
> On our module we have external mux switching between the e.MMC (on the module)
and external SDcard socket, very similar to Icicle board, but controlled via MSS GPIO 12.
At the moment we have modified the code to work with our hardware and this
issue is "just for the records", to keep track of what we think may be improved.

Cool 😄 thats interesting - the reason we added the register (we had previously used the GPIO) was that when Linux came up and initialized the GPIO driver it would also initialize all of the GPIOs to a value of 0, grand if you're using eMMC but if you're using SD this would unmount the card, and just to be confusing, it would also manage to mount the eMMC and boot rootfs from eMMC. This is hard to spot as everything looks OK but in reality you're running Linux loaded from SD using a rootfs from eMMC. To solve this we moved to the memory mapped register from the GPIO as you could still cause this to happen but you'd have to be a bit more deliberate in doing it and Linux won't touch the register when it boots up. 

I'll update this issue once the feature has been released in Libero so you get a heads up, thanks for pointing it out, we did this when we didn't have many partner boards so we should probably keep an eye on this for the different configurations starting to pop up.

ak-at-aries, 2021-03-16:
> This is hard to spot as everything looks OK but in reality you're running Linux loaded from SD using a rootfs from eMMC. 

Yea... can imaging how confusing it is.

Using an FPGA block and pin doesn't look to us attractive: it would place an extra restriction on the custom FPGA project, waste an FPGA pin (while the GPIO we have for "free"), and is likely to cause the same confusing behavior you had, only not for us, but for our customers, which would be worse.  The current plan is to address the problem in the Linux GPIO driver (or, hopefully, just in the device tree).

Anyway, thank you for pointing out the tricky problem with the mux and the Linux GPIO driver.

Do you know if the IOMUXx_CR register fields values are documented somewhere?
All we were able to find is just a reference to MSS MAS, whatever it is.

I guess this issue may be closed now, or just snoozed till mss configuration software implements required functionality.

hughbreslin, 2022-11-03:
This has been updated in our latest release to support switching based on the XML and is available now, you can see the changes [here](https://github.com/polarfire-soc/hart-software-services/blob/master/services/mmc/mmc_api.c#L113). I hope this is sufficient to close this issue

vfalanis, 2023-03-16:
Closing this issue due to lack of activity. This functionality has been updated and published on GitHub. If you need additional information please reopen this issue.

Thanks

closed 2023-03-16 by vfalanis