polarfire-soc/hart-software-services #19: HSS does not compile using the gcc version 10.2.0

lbmeng, 2021-03-19:
Use the pre-built toolchain from
https://toolchains.bootlin.com/downloads/releases/toolchains/riscv64/tarballs/riscv64--glibc--bleeding-edge-2020.08-1.tar.bz2

On the following HSS commit: (HEAD as of now)
```
commit 58b03943834fe34991dc5fa924436b3620e07aa5 (HEAD -> master, origin/master, origin/HEAD)
Merge: 7383f7e c3341b9
Author: Cyril-Jean <cyril.jean@microchip.com>
Date:   Tue Jan 19 10:43:27 2021 +0000

    Merge pull request #10 from kraj/master
    
    hss-payload-generator: Rename EM_ARC_COMPACT2 to EM_ARCV2
```

$ export CROSS_COMPILE=riscv64-linux-
$ make BOARD=mpfs-icicle-kit-es

```
riscv64-buildroot-linux-gnu/bin/ld: Default/hss.elf section `.text' will not fit in region `envm'
riscv64-buildroot-linux-gnu/bin/ld: region `envm' overflowed by 9792 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:177: hss.elf] Error 1
```

lbmeng, 2021-03-19:
Found a workaround:

`$ export CONFIG_WITH_ARCH=1`

But this whole CONFIG_WITH_ARCH is undocumented and we should not let user to configure this.
We should explicitly set our build flags so that HSS does not depend on the toolchain pre-configured settings.

griffini, 2021-03-19:
CONFIG_WITH_ARCH went in to work around some Linux cross compiler issues from a while back - see https://github.com/riscv/riscv-pk/pull/114. As a result of this, we truly only support and test against the compiler bundled with Microchip SoftConsole. While other toolchains can work, they may indeed have certain issues at any particular point in time.  

We can remove the CONFIG_WITH_ARCH patch again now - it was only ever a temporary workaround to build problems with Yocto, but we've since moved the HSS out of Yocto.

The larger problem here is that the HSS (and in particular OpenSBI) has grown beyond the available space in eNVM. To that end, our proposed solution that we are currently testing is to compress the HSS image in eNVM, and to then decompress it to the L2 LIM/scratchpad at startup.

lbmeng, 2021-03-19:
> CONFIG_WITH_ARCH went in to work around some Linux cross compiler issues from a while back - see [riscv/riscv-pk#114](https://github.com/riscv/riscv-pk/pull/114). As a result of this, we truly only support and test against the compiler bundled with Microchip SoftConsole. While other toolchains can work, they may indeed have certain issues at any particular point in time.
> 
> We can remove the CONFIG_WITH_ARCH patch again now - it was only ever a temporary workaround to build problems with Yocto, but we've since moved the HSS out of Yocto.

Yes, this CONFIG_WITH_ARCH workaround should be removed now. Please check my PR.

> The larger problem here is that the HSS (and in particular OpenSBI) has grown beyond the available space in eNVM. To that end, our proposed solution that we are currently testing is to compress the HSS image in eNVM, and to then decompress it to the L2 LIM/scratchpad at startup.

This is an independent feature and could help on the larger size problem. But the issue reported here should be fixed anyway.

griffini, 2021-03-30:
We're in the final stages of some restructuring of the HSS code that will make your PR redundant (it won't apply), but we'll definitely be passing the arch info when building all source files.

lbmeng, 2021-03-30:
> We're in the final stages of some restructuring of the HSS code that will make your PR redundant (it won't apply), but we'll definitely be passing the arch info when building all source files.

But the restructuring is irrelevant with this issue. IMHO the PR should be merged to resolve this issue for any downstream project that might not be moving to the future "restructured" repo.

griffini, 2021-04-12:
> But the restructuring is irrelevant with this issue. IMHO the PR should be merged to resolve this issue for any downstream project that might not be moving to the future "restructured" repo.

Okay, let's merge it in for now.  It is being restructured, and the restructuring has some other fixes that downstream projects should consider, but as you correctly say, this is a separate concern.

closed 2021-04-19 by Cyril-Jean