polarfire-soc/hart-software-services #21: HSS not compiling under softconsole- following "Training #4: Hart Software Services"

rroulleau, 2021-03-22:
After following training #4 from :
https://www.microchip.com/training/webinars/fpgas/risc-v-innovation-unleashed-training-series/risc-v-innovation-unleashed-training-archive
I followed the steps up to making the build but I am having the followin error:
"
16:09:57 **** Incremental Build of configuration Default for project hart-software-services ****
env MSYSTEM=MSYS2 "PATH=/bin:/usr/bin:/usr/local/bin:/home//.local/bin:C:\\Microchip\\SoftConsole-v6.6\\/riscv-unknown-elf-gcc/bin" BOARD=mpfs-icicle-kit-es make all 
Cannot run program "env": Launching failed

Error: Program "env" not found in PATH
"
Is the gitlab out of synch with softconsole? 
Also I tried to compile it via linux but get into trouble with the  cross compiler  riscv64-linux-gcc, I did not find which file to source to setup the environment variables. Or do I need to set them manually?

hughbreslin, 2021-03-22:
Hey @rroulleau did you add msys2 to your system path as shown in step 4 [here](https://github.com/polarfire-soc/polarfire-soc-documentation/blob/master/software-development/polarfire-soc-software-tool-flow.md#config-msys2)? This could explain the error you're seeing.

If you try to build from the command line (either from msys2 or Linux) you'll need to manually point to a RISC-V toolchain using your environment variables.

rroulleau, 2021-03-23:
Obviously not! :-/ thanks, it is now ok on eclipse. As a thought would it be worth to add a section pre-requisite on the hart-software-services readme with a link to the step 4 you mentioned? For a newcomer any little help count :-)

closed 2021-03-23 by rroulleau

hughbreslin, 2021-03-23:
Yes good point I'll try and get the readme updated with a link to the set up steps in this doc :)

gege0500, 2022-01-19:
@hughbreslin How can I add msys2 to my system path? I didn't find it in the step 4 you mentioned.

Shivaraj199, 2022-04-25:
Do i need to configure softconsole for linux,
The make has been executed sucessfully and i can see generated files in default directory. but not hss.bin

gege0500, 2022-04-25:
收到

griffini, 2022-04-26:
@Shivaraj199, you need to configure your Linux shell to setup environment variables SC_INSTALL_DIR to point to your SoftConsole install, and FPGENPROG to point to the fpgenprog binary of your Libero installation.

For example
````
export SC_INSTALL_DIR=/opt/microchip/SoftConsole-v2021.3-7.0.0.578
export FPGENPROG=/opt/microchip/Libero_SoC_v2021.2/Libero/bin64/fpgenprog
````

When you build the HSS, in the `Default` directory you should then see:
* `hss-envm-wrapper.hex` - this is your binary file suitable for adding to your Libero project and programming  (or via the External Tools: "PolarFire SoC program non-secure boot mode 1"  option in SoftConsole, or by running `make program` from the HSS toplevel directory)
* `hss-l2scratch.elf` - this is an ELF file suitable for loading via the debugger.

Shivaraj199, 2022-04-27:
@griffini Thank you, I have pointed to the softconsole directory and build. After building the HSS this are the files I get in Default directory.
hss-envm-wrapper.bin  hss-l2scratch.bin.deflate  output-l2scratch.map
hss-envm-wrapper.elf  hss-l2scratch.elf          README.md
hss-envm-wrapper.sym  hss-l2scratch.sym
hss-l2scratch.bin     output-envm-wrapper.map

After I run the command 
$ qemu-system-riscv64 -M microchip-icicle-kit -smp 5 \
    -bios path/to/hss-envm-wrapper.bin -sd path/to/sdcard.img \
    -nic user,model=cadence_gem \
    -nic tap,ifname=tap,model=cadence_gem,script=no \
    -display none -serial stdio \
    -chardev socket,id=serial1,path=serial1.sock,server=on,wait=on \
    -serial chardev:serial1

This is the output I get in the terminal 
qemu-system-riscv64: -chardev socket,id=serial1,path=serial1.sock,server=on,wait=on: info: QEMU waiting for connection on: disconnected:unix:serial1.sock,server=on

when i open the minicom in another terminal, there is no output.
where am i am going wrong?

Shivaraj199, 2022-05-09:
Hello thanks for the reply,
I am trying to emulate icicle kit on qemu. As the qemu page is obsolete I
cant make any use of it. I tried building image using yocto but I saw yocto
dosnt support qemu platform.  Can you please specify the steps how can I
emulate the board on polarfire?

Thank you,