polarfire-soc/hart-software-services #22: Board hangs at selecting SDCARD

eenurkka, 2021-04-09:
With the latest hart-software-services as of today (commit: 58b03943834fe34991dc5fa924436b3620e07aa5), the sdcard detection hangs always:

[1.939392] HSS_MMCInit(): Attempting to select SDCARD ... ▒▒▒

Commenting this line below will no longer cause the board to hang:
HW_set_uint32(SDIO_REGISTER_ADDRESS,  1);

but the sdcard never works. I've tried 3.3v/1.8v dip switch, different releases, no matter what, the system gets an error at MSS_MMC_init() function, on line 657. I'm using the board mpfs-icicle-kit-es and Microchip SoftConsole v6.5.0.442.

ziqiaozhou, 2021-08-19:
I got the same issue. The board hangs when init SDcard. 
HSS_MMCInit(): Attempting to select SDCARD ... ▒▒▒
Then the problem disappear after unsetting CONFIG_SERVICE_SDIO_REGISTER_PRESENT in config (skip HW_set_uint32(SDIO_REGISTER_ADDRESS, 1)). My SD card never works. I have no idea whether it is a hardware problem or the software problem.

ziqiaozhou, 2021-08-19:
> With the latest hart-software-services as of today (commit: [58b0394](https://github.com/polarfire-soc/hart-software-services/commit/58b03943834fe34991dc5fa924436b3620e07aa5)), the sdcard detection hangs always:
> 
> [1.939392] HSS_MMCInit(): Attempting to select SDCARD ... ▒▒▒
> 
> Commenting this line below will no longer cause the board to hang:
> HW_set_uint32(SDIO_REGISTER_ADDRESS, 1);
> 
> but the sdcard never works. I've tried 3.3v/1.8v dip switch, different releases, no matter what, the system gets an error at MSS_MMC_init() function, on line 657. I'm using the board mpfs-icicle-kit-es and Microchip SoftConsole v6.5.0.442.

I also encountered the same problem today. I just figured out the reason after reading their release comments. It is because the board's FPGA design is not compatible to support that SDIO_REGISTER. My board design is the original one. The original design does not support dynamic switching between SD card and eMMC.  After programming the lastest [icicle kit reference design](https://github.com/polarfire-soc/icicle-kit-reference-design.git) to my board using Libero, it now can boot well using the latest HSS. :>

hughbreslin, 2021-08-20:
Glad to hear this is working for you now 🙂

closed 2021-09-23 by griffini