polarfire-soc/hart-software-services #29: [tag 2021.08] Error on SC "Run PolarFire SoC program non-secure boot mode 1"

okawaj9, 2021-08-15:
[Error message on SoftConsole v2021.1]
ERROR - ELF file linked incorrectly. The boot vector for the eNVM must be between "0x20220000" and "0x2023FFFF" but the selected ELF file "hss-l2lim.elf" has boot vector "0x08000000". Check the linker script used to link the program.

[Operation]
* Import HSS (master or tag=2021.08) into SC v2021.1
* Build (default) on SC v2021.1
* (program PF SoC reference design - 2021.08 into ICICLE-KIT)
* On SC v2021.1, "Run PolarFire SoC program non-secure boot mode 1" along HSS Youtube video

[Inquiry]
From 2021.08, what addition operation is required to program HSS into eNVM through SC v2021.1 ?

griffini, 2021-09-02:
Hi @okawaj9,

Are you running SoftConsoel on Linux or on Windows ?

It sounds like the wrong ELF file is getting picked up in programming.  We will investigate this -  as a workaround for now, try deleting the file "Default/hss-l2lim.elf" after you have compiled but befure you attempt programming.

okawaj9, 2021-09-02:
Hi,

Thank you for your reply.  It was on Ubuntu 20.04.2 LTS (see attachment).

$ uname -a
Linux xxx 5.11.0-27-generic #29~20.04.1-Ubuntu SMP Wed Aug 11 15:58:17 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Today, I reproduced this issue again by following operations with SC v2021.1
1. Download HSS zip file from this github after setting tag=2021.08 (filename: hart-software-services-2021.08.zip)
2. Import the zip file into SC v2021.01 workspace
3. chmod +x [workspace]/hart-software-services/thirdparty/Kconfiglib/genconfig.py
4. Copy "hart-software-services/boards/mpfs-icicle-kit-es/def_config" to "hart-software-services/.config"
5. Build HSS (build target=Default)
6. Select "hart-software-services" on Projet Explorer on SC
7. Run Polarfire SoC Program non-secure boot mode 1 (see attached picture)

Best Regards,

[image]

[image]

griffini, 2021-11-25:
@okawaj9, delighted it is working for you.  This problem with the second ELF is actually now a known tools issue that will be fixed in an upcoming version of SoftConsole. I hope it is okay if I close this as it isn't a HSS specific issue, and we have a workaround.

closed 2021-11-25 by griffini