polarfire-soc/hart-software-services #31: mss_sgmii: likely typo?

eenurkka, 2021-09-07:
In the file: baremetal/polarfire-soc-bare-metal-library/src/platform/mpfs_hal/common/nwc/mss_sgmii.c:

There's a duplicate line, should the other be MSS_PERIPH_MAC1 instead of MSS_PERIPH_MAC0?

            (void)mss_config_clk_rst(MSS_PERIPH_MAC0, (uint8_t) MPFS_HAL_FIRST_HART, PERIPHERAL_ON);
            (void)mss_config_clk_rst(MSS_PERIPH_MAC0, (uint8_t) MPFS_HAL_FIRST_HART, PERIPHERAL_ON);
            GEM_A_LO->network_config |= (0x01U << 10U) | (0x01U << 11U);   /* GEM0 */
            GEM_B_LO->network_config |= (0x01U << 10U) | (0x01U << 11U);   /* GEM1 */

eenurkka, 2021-09-07:
This seems to be writing in the NETWORK_STATUS register (which is RO), although one would expect a write in the NETWORK_CONFIG  -register?

            GEM_A_LO->network_config |= (0x01U << 10U) | (0x01U << 11U);   /* GEM0 */
            GEM_B_LO->network_config |= (0x01U << 10U) | (0x01U << 11U);   /* GEM1 */

mss_sgmii.h file declares the base 4 bytes ahead of what's on the specs (ought to be 0x20110000 and 0x20112000 instead?):

#define GEM_A_LO_BASE  0x20110004UL
#define GEM_B_LO_BASE  0x20112004UL

typedef struct IOSCB_GEM_X_LO_STRUCT_ {
    __IO uint32_t network_control;      /* The network configuration register contains functions for setting the mode of operation for the Gigabit Ethernet MAC.  */
    __IO uint32_t network_config;       /*             */
    __IO uint32_t network_status;       /*             */
    __IO uint32_t dma_config;           /*             */
    __IO uint32_t transmit_status;      /*             */
    __IO uint32_t receive_q_ptr;        /*             */
} IOSCB_GEM_X_LO_STRUCT;

#define GEM_A_LO  ((volatile IOSCB_GEM_X_LO_STRUCT *) GEM_A_LO_BASE)
#define GEM_B_LO  ((volatile IOSCB_GEM_X_LO_STRUCT *) GEM_B_LO_BASE)

griffini, 2021-09-07:
Thanks for reporting these.