polarfire-soc/hart-software-services #33: HSS fails to detect SDCARD and eMMC (release 2021-08)

cmey, 2021-09-20:
Hi,

I am evaluating the Icicle kit, board Rev 1.0.
I flashed [release 2021-08](https://github.com/polarfire-soc/icicle-kit-reference-design/releases/tag/2021.08) (using FlashProExpress, directly from the prebuilt .job file). The flashing process completes to 100% without error and I also ran a VERIFY step which completes to 100% with no error.

I am connected to COM0 and COM1.  
**The board starts up but HSS fails to detect both the SDCARD and the eMMC.**

Here is the complete log from COM0:
```
[7.212622] HSS_E51_Banner(): PolarFire(R) SoC Hart Software Services (HSS) - version 0.99.23                                                                            
MPFS HAL version 1.8.131                                                                                                                                                
(c) Copyright 2017-2021 Microchip FPGA Embedded Systems Solutions.                                                                                                      
                                                                                                                                                                        
[7.230937] HSS_E51_Banner(): incorporating OpenSBI - version 0.9                                                                                                        
(c) Copyright 2019-2021 Western Digital Corporation.                                                                                                                    
                                                                                                                                                                        
[7.243349] HSS_PrintBuildId(): Build ID: 209830aa388166b34cb2876fde1f2b48990fec7f                                                                                       
[7.252376] HSS_PrintToolVersions(): Built with the following tools:                                                                                                     
 - riscv64-unknown-elf-gcc (xPack GNU RISC-V Embedded GCC (Microsemi SoftConsole build), 64-bit) 8.3.0                                                                  
 - GNU ld (xPack GNU RISC-V Embedded GCC (Microsemi SoftConsole build), 64-bit) 2.32                                                                                    
                                                                                                                                                                        
[7.276854] HSS_E51_Banner(): NOTICE: Running from L2LIM                                                                                                                 
                                                                                                                                                                        
[7.283798] HSS_DDRPrintSegConfig(): Segment Configuration:                                                                                                              
        Cached: SEG0_0: offset 0x0080000000, physical DDR 0x00000000                                                                                                    
        Cached: SEG0_1: offset 0x1000000000, physical DDR 0x30000000                                                                                                    
    Non-cached: SEG1_2: offset 0x00c0000000, physical DDR 0x70000000                                                                                                    
Non-cached WCB: SEG1_4: offset 0x00d0000000, physical DDR 0x70000000                                                                                                    
[7.315133] HSS_DDRPrintL2CacheConfig(): L2 Cache Configuration:                                                                                                         
    L2-Scratchpad:  4 ways (512 KiB)                                                                                                                                    
         L2-Cache:  8 ways (1024 KiB)                                                                                                                                   
           L2-LIM:  4 ways (512 KiB)                                                                                                                                    
[7.332580] HSS_DDRPrintL2CacheConfig(): L2 Cache Way Masks:                                                                                                             
              DMA: 0xf0ff                                                                                                                                               
AXI4_SLAVE_PORT_0: 0xf0ff                                                                                                                                               
AXI4_SLAVE_PORT_1: 0xf0ff                                                                                                                                               
AXI4_SLAVE_PORT_2: 0xf0ff                                                                                                                                               
AXI4_SLAVE_PORT_3: 0xf0ff                                                                                                                                               
       E51_ICACHE: 0xf0ff                                                                                                                                               
     U54_1_DCACHE: 0xf0ff                                                                                                                                               
     U54_1_ICACHE: 0xf0ff                                                                                                                                               
     U54_2_DCACHE: 0xf0ff                                                                                                                                               
     U54_2_ICACHE: 0xf0ff                                                                                                                                               
     U54_3_DCACHE: 0xf0ff                                                                                                                                               
     U54_3_ICACHE: 0xf0ff                                                                                                                                               
     U54_4_DCACHE: 0xf0ff                                                                                                                                               
     U54_4_ICACHE: 0xf0ff                                                                                                                                               
       E51_DCACHE: 0xf0ff                                                                                                                                               
[7.374851] HSS_MemTestDDRFast(): DDR-Lo size is 768 MiB                                                                                                                 
[7.381361] HSS_MemTestDDRFast(): DDR-Hi size is 1 GiB                                                                                                                   
[7.387698] HSS_BoardLateInit(): Please ensure that jumpers J34/J43 are correct for 1.8V MMC voltage...                                                                  
[7.398634] HSS_MMCInit(): Attempting to select SDCARD ... Failed                                                                                                        
[8.409828] HSS_MMCInit(): Attempting to select eMMC ... Failed                                                                                                          
[9.420450] RunInitFunctions(): HSS_MMCInit() returned 0                                                                                                                 
Press a key to enter CLI, ESC to skip                                                                                                                                   
Timeout in 1 second                                                                                                                                                     
[... cut ...]
I have an SDCARD inserted.  I also tried a brand new SDCARD, same issue.

I found out that if I flash the older release v2021-02, then after turning off and on the board about 5 times, it finally detects the SDCARD. After that initial "warm up" phase, it then detects the SDCARD about half of the time. The eMMC however is never detected, in any version that I tried (v2021-02 to v2021-08).

I would like the eMMC and the SDCARD to be detected, so that I can boot Linux from it.
Any help would be appreciated!

hughbreslin, 2021-09-21:
Hi @cmey 🙂 sorry you're having this issue! 

Can I check a few things:
1. In your partially working configuration using an older design, does it work at both 1.8v and 3.3v or just 3.3v? 
2. Was your board booting OK out of the box and did these issues only show up when you updated the design or have you always had an issue booting from the eMMC / SD cards?
2. Can you check what setting J35 (VDDAUX4) is in? Its between the PolarFire SoC part and the FlashPro header. Could you try setting it to 2 & 3 closed (2.5v) when using the latest design with J34 and J43 set to 1.8v?

cmey, 2021-09-21:
Thank you for getting back to me so quickly!

> 1. In your partially working configuration using an older design, does it work at both 1.8v and 3.3v or just 3.3v?

I just tested that now and, v2021-02 works in both 1.8v and 3.3v, i.e. jumpers J34 and J43 at 1&2 (both jumpers together) or 2&3 (both jumpers together) both work. Work in the sense that the SD CARD is detected.
In fact, it may be just a good string of random luck, but it seems to work better at 1.8v, i.e. it hasn't failed to detect the SD CARD yet with v2021-02 with J34 and J43 at 1.8v (2&3 - right most - connected together).
It still never succeeds to detect the eMMC.

> 2. Was your board booting OK out of the box and did these issues only show up when you updated the design or have you always had an issue booting from the eMMC / SD cards?

Board booted OK out of the box, issues started after I updated.   I do not know what version the board came with.

> 3. Can you check what setting J35 (VDDAUX4) is in? Its between the PolarFire SoC part and the FlashPro header. Could you try setting it to 2 & 3 closed (2.5v) when using the latest design with J34 and J43 set to 1.8v?

J35 is at 2&3 (I think, it's in the position where the bottom 2 pins are connected together).  It has always been there, so I have tested that configuration - that is the configuration 2021-08 reported above.

Thank you for looking into this :pray:

hughbreslin, 2021-09-23:
Hey @cmey

> J35 is at 2&3 (I think, it's in the position where the bottom 2 pins are connected together). It has always been there, so I have tested that configuration - that is the configuration 2021-08 reported in the original post.

I'm not sure which way is up for your kit to know what the bottom pins are for J35, the pin with the arrow pointing at it should be open. This wasn't the original configuration on some kits.

If its not that, it sounds like it could be some kind of board issue, we aren't seeing the same behavior on our end with kits running 2021.08. Would you be able to open a tech support case with your supplier to try and see about getting a replacement?

cmey, 2021-09-23:
I'm looking at the board in the way that I can read "Microchip ICICLE Kit Rev: 1.0" on the bottom right of the board.

Yes, J35 is set in a way that the pin pointed to by the arrow above it is open, and the other, bottom 2 pins, are closed.

OK, we have a second identical board that arrived here that I will test.

hughbreslin, 2021-09-23:
Ok sounds good, hopefully theres no issue with the second board - please let me know!

cmey, 2021-10-08:
Confirming that the new replacement board works, both with the software that it came with and also after updating to 2021-08.
Thanks!

closed 2021-10-08 by cmey