polarfire-soc/hart-software-services #34: Generate bitstream failed

luoflyxidian, 2021-09-23:
When I am going to burn HSS through boot mode 1, I encountered the following problem.

[ Generate bitstream... ]
The eNVM configuration has the following validation errors:
The ENVM configuration has the following errors:
    **Client 'bootmode1': Unable to open memory file 'hss-envm-wrapper.hex'.**
Error: Failed to generate eNVM EFC from CFG. Return code = 1
Error:   generate bitstream failed, exiting...
Error:   check the log for errors and refer to D:/Microchip/SoftConsole-v6.5//extras/mpfs/readme.txt if necessary
[ Finished (failure) ]

I have checked the path of fpgenprog.exe. Has anyone encountered such a problem?

griffini, 2021-09-23:
Hi @luoflyxidian,

Sorry that you're having a difficulty programming. 

Does the file `Default/hss-envm-wrapper.hex` exist?

What do you have the FPGENPROG environment variable set to?

luoflyxidian, 2021-09-28:
The file Default/hss-envm-wrapper.hex does exist.  

In softconsole.cmd, I set FPGENPROG path as follows.
if not defined FPGENPROG set FPGENPROG=D:\Microsemi\Libero_SoC_v2021.1\Designer\bin64\fpgenprog.exe

In addition, boot mode 0 is available.

Cyril-Jean, 2021-12-02:
Don't you need to use 'make' first to build the HSS followed by 'make program' to perform the programming? As in, 'make program' on its own does not build the HSS.

griffini, 2021-12-02:
Yes, it must be built first -- either `make` first from command line to build and then `make program` to prepare and program the HSS image, or else build using Soft Console and then use the External tools option to run "Polarfire Soc program non-secure boot mode 1"

I wasn't clear whether @luoflyxidian was using command line or Soft Console, but one potential explanation for the issue is that the HSS might not have been built first?

griffini, 2021-12-16:
@luoflyxidian, did you manage to get things to work, or are you still encountering problems?

griffini, 2022-02-17:
Closing this as stale.

closed 2022-02-17 by griffini