polarfire-soc/hart-software-services #37: Custom OpenSBI runs only on one core

neiling, 2021-12-01:
Sorry if this is not a bug, just maybe a misconfiguration.

My use case is to run the [Keystone Enclave](https://github.com/keystone-enclave/keystone) on the PolarFire SoC, for
this, I need a custom OpenSBI as [security monitor](https://github.com/keystone-enclave/sm). 

I builded the OpenSBI firmware with the [U-Boot](https://u-boot.readthedocs.io/en/latest/board/microchip/mpfs_icicle.html) as payload.

In principle everything works fine but during the boot process the Linux kernel
can't access the other cores. I have the same problem when using the vanilla
OpenSBI firmware.

Do I need to change anything in the default HSS config if I want to use my own
OpenSBI?

griffini, 2021-12-01:
You want to make sure that you boot your payloads in M-mode, and also use the skip-opensbi flag in the payload generator YAML as described in https://github.com/polarfire-soc/hart-software-services/blob/master/tools/hss-payload-generator/README.md ...

E.g., something like ...

```
set-name: 'CustomOpenSBI'
hart-entry-points: {u54_1: '0x80200000', u54_2: '0x80200000', u54_3: '0x80200000', u54_4: '0x80200000'}
payloads:
  test/u-boot-dtb.bin:    {exec-addr: '0x80200000', owner-hart: u54_1, secondary-hart: u54_2, secondary-hart: u54_3, secondary-hart: u54_4, priv-mode: prv_m, skip-opensbi: true}
```

neiling, 2021-12-01:
Hi griffin,

thanks for your answer.

I mean I already do that. The strange thing is if I use the skip-opensbi flag then I get CPU{0,2} online but not CPU{1,3}.

```
set-name: 'PolarFire-SoC-HSS::SM_U-Boot'
hart-entry-points: {u54_1: '0x80000000', u54_2: '0x80000000', u54_3: '0x80000000', u54_4: '0x80000000'}
payloads:
  fw_payload.elf: {exec-addr: '0x80000000', owner-hart: u54_1, secondary-hart: u54_2, secondary-hart: u54_3, secondary-hart: u54_4, priv-mode: prv_m, skip-opensbi: true}
```

griffini, 2021-12-01:
I am assuming all your code is running from DDR and not L2LIM? 

If U54_1 and U54_3 are coming up and not U54_2 and U54_4, what code are U54_2 and U54_4 running? Is it in your custom OpenSBI?

If not, can you put a breakpoint (ebreak) in at 0x80000000 to allow each hart to trap to the debugger at startup?  If all harts are trapping, then the HSS has successfully handed over control.

neiling, 2021-12-02:
Sorry, I am not so familiar with this kind of debugging.

I tried the openocd tool, but I'm not sure if it's the right one. I haven't figured out how to set a breakpoint to a specific hart, whenever I connect to gdb I end up on the hart0_e51.

Furthermore, I played again with the YAML config, and noticed that the problem really only occurs when I want to use more than two cores.
According to your assumption, I would expect if I specify only two cores that the Linux kernel can then allocate only one and on the other runs the OpenSBI, but I get both.

griffini, 2021-12-09:
If you allocate only two cores via the payload generator, Linux will still try to grab up to as many as are enabled in its device tree. So you may have to edit the device tree to disable certain cores.

Regarding GDB debugging, if you connect via GDB, you can set the hart you want to look at with

`(gdb) thread x`

where x is 1 to 5. 1 is E51, 2 is U54_1, etc.

You can also see where all threads are with

`(gdb) info threads`

griffini, 2021-12-16:
Hi @neiling, I am just wondering if you have managed to make any progress with this or if you are still having an issue?

neiling, 2021-12-17:
Hi griffini, thank you very much for the help.
I have now tried again to debug and as far as I can see the Opensbi runs only on the cores on which the Linux runs. However, what runs or happens on the others I could not find out until now. I never managed to trigger a breakpoint there. As you can see in the screenshot, the two u54_{1,4} usually remain at 0x800c9XX.

[image]

Linciclon, 2021-12-17:
Same issue here, even if I put "skip-opensbi: true", the OpenSBI logger appears anyway...
Currently I'm using:
'payloads:
<path to opensbi >/opensbi/build/platform/generic/firmware/fw_payload.bin: {exec-addr: '0x80000000', owner-hart: u54_1, priv-mode: prv_m, skip-opensbi: true}`

I'm using the pre-buit hss-payload-generator from "Hart Software Services Release 2021.11" release, to get payload.bin file.

And as output, I'm getting 2 OpenSBI's loggers :)
However, if I add more harts to config.yaml file none of OpenSBI print the logger....

griffini, 2022-05-05:
HI @Linciclon , if you can could you retest with the 2022.03 release? There were a number of small bugs resolved that may have been affecting your use case?

griffini, 2022-07-07:
Hi @Linciclon, @neiling, I'm going to close this as the issue has been idle for some time and, as I mentioned, some relevant bug fixes in 2022.03 should have resolved any issues with booting bare-metal projects in M mode.

If you still have issues, please do let us know and we'll investigate.

closed 2022-07-07 by griffini

rejsys, 2023-03-01:
> Hi griffin,
>
> thanks for your answer.
>
> I mean I already do that. The strange thing is if I use the skip-opensbi flag then I get CPU{0,2} online but not CPU{1,3}.
>
> ```
> set-name: [...]

I have got the same issue with my setup. There is another [open issue](https://github.com/polarfire-soc/polarfire-soc-documentation/discussions/38) related to this thread. I found out that the `mhartid` CSR field does not hold the right HART ID but that is all I could find out yet.

EmdaloTechnologies, 2023-03-02:
Hi @rejsys 

Can you provide a bit more info on what you are trying to do? I don't understand what you mean by `mhartid` not holding the right ID.

Which U54s are you trying to start? And can you see if they start and have fatally trapped? Are you able to get mcause or back traces?

rejsys, 2023-03-02:
Hello back,

I used a similar setup to start an application on hart 1 which than interrupts the other harts. In fact, u-boot is the hss-payload and is configured the same as above `owner-hart: u54_1, secondary-hart: u54_2, secondary-hart: u54_3, secondary-hart: u54_4`.
I found out that uboot has an internal "HART lottery system" which (even though the owner hart is u54_1) runs on u54_2 and interrupts the other cores.