polarfire-soc/hart-software-services #41: HSS fails to boot on MPFS250T_ES FCSG536

fcuzzocrea, 2022-03-03:
Ciao!

This is just to report that, current version of HSS fails to boot on MPFS250T_ES die FCSG536 package. The issue seems to be somewhere into the HAL code.

From my tests, something in the mss_nwc_init() causes the CPU to reset. To get it working, I had to update the mpfs_hal to the latest version available from the platform repo.

Do you plan to update BsP version shipped with HSS ? Thanks

Cyril-Jean, 2022-03-03:
Interesting. Could you try with the current master of the HSS?
Which type of DDR are you using? LPDDR4 or other?

fcuzzocrea, 2022-03-03:
No, simply DDR4. I can give a shot to the current master later and let you know!

fcuzzocrea, 2022-03-03:
I just had a try - current master of HSS boots fine

griffini, 2022-03-08:
> I just had a try - current master of HSS boots fine

@fcuzzocrea , delighted to hear this. I'll close this issue now.

closed 2022-03-08 by griffini