polarfire-soc/hart-software-services #42: No rule to make target 'baremetal/drivers/mss_envm/mss_envm.c' in case of CONFIG_SERVICE_YMODEM=y

okawaj9, 2022-03-17:
1. Issue:
HSS build error in case of CONFIG_SERVICE_YMODEM=y for conducting PMP demo

> make: *** No rule to make target 'baremetal/drivers/mss_envm/mss_envm.c', needed by 'baremetal/drivers/mss_envm/mss_envm.o'.  Stop.

2. hart-software-services / boards / mpfs-icicle-kit-es / Makefile
EXTRA_SRCS-$(CONFIG_SERVICE_YMODEM) += baremetal/drivers/mss_envm/mss_envm.c

3. "baremetal/drivers/mss_envm" not found:
https://github.com/polarfire-soc/hart-software-services/tree/master/baremetal/drivers

Could you please release **baremetal/drivers/mss_envm** for PMP demo, please ?

Best Regards,

nitindeshpande, 2022-03-17:
Hi,
It was decided to remove the eNVM driver source code from the HSS, but there is
a bug that it is included in a Makefile when the YMODEM option is enabled.

Could you try this workaround for now:
In boards/mpfs-icicle-kit-es/Makefile, remove the following lines:

EXTRA_SRCS-$(CONFIG_SERVICE_YMODEM) += \
       baremetal/drivers/mss_envm/mss_envm.c

We'll make this fix in the HSS source repo.

okawaj9, 2022-03-17:
Thank you for your reply.
 
> It was decided to remove the eNVM driver source code from the HSS, but there is
> a bug that it is included in a Makefile when the YMODEM option is enabled

Then could you please also update PMP demo instruction described in https://github.com/polarfire-soc/polarfire-soc-documentation/blob/master/demo-guides/mpfs-pmp-demo.md, please ?

It requires HSS build from source code with CONFIG_SERVICE_YMODEM=y, and use YMODEM feature with HSS.

(The workaround you introduced helps to get "pass" of HSS build.  But it does not help PMP demo flow described in   https://github.com/polarfire-soc/polarfire-soc-documentation/blob/master/demo-guides/mpfs-pmp-demo.md)

Best Regards,

griffini, 2022-04-26:
Closing this issue as baremetal/drivers/mss_envm/mss_envm.c is no longer referenced by the HSS build system as of [commit bfc5555e811627caa599e5b2fe4c7b6d32272cb6](https://github.com/polarfire-soc/hart-software-services/commit/bfc5555e811627caa599e5b2fe4c7b6d32272cb6)

closed 2022-04-26 by griffini