polarfire-soc/hart-software-services #50: source of ./tools/hss-payload-generator/test/baremetal.elf missing

xypron, 2022-08-19:
Packaging the hss-payload-generator for a Linux distro should include all source code needed to create the distributed binaries.

Could you, please, make the source of baremetal.elf available? Should be a file helloWorld.c.

Ideally you would add a Makefile target to cross compile it.

vfalanis, 2022-09-01:
Hi @xypron,

We pushed an update to the master branch to remove the elf binary from the test directory, since it shouldn't have been pushed there.

For anyone interested in booting a bare metal elf file from the HSS, we recommend to build any of the bare metal example applications provided in the polarfire-soc-bare-metal-examples [repository](https://github.com/polarfire-soc/polarfire-soc-bare-metal-examples.git) instead.

Thanks

xypron, 2022-09-01:
Thanks for resolving the issue.

closed 2022-09-01 by xypron