polarfire-soc/hart-software-services #52: HSS struggling to reboot the system on watchdog timeout

nearly-big-endian, 2022-11-15:
Hi,

We are relying on the watchdog mechanism to have the board rebooted in case of system freeze (Linux in our case).

That feature used to work as expected on the HSS 0.99.26 (together with Reference design 2021.11 and Polarfire Yocto BSP 2021.11) on an IcicleKit, in the sense that the board did get successfully restarted on watchdog timeout (in case it was not refreshed).

We however noticed since upgrading to HSS 0.99.33 // Ref Design 2022.09 // Yocto BSP 2022.09 that HSS fails to restart the board on first watchdog timeout signal.

It eventually manages to restart the board, but only after having received 9 more watchdog timeout signals (as the log shows below). If the watchdog runtime duration is set to 30 seconds and the watchdog is not actively refreshed, the board therefore ends up being effectively rebooted only after about 300 seconds instead of 30 seconds (that is, at the end of 10 watchdog timeout signals).

Could it be a regression in the HSS ?

Here is a set of Linux commands that may be used to reproduce the issue, using the devmem2 linux command line tool. 
This sets up the U54 watchdog 1 with the maximum runtime duration (about 30 seconds) and let it expire.
The expected outcome is that the board gets rebooted immediately on watchdog timeout.

```bash
# -- Setup and start watchdog 1 --
# Set WATCHDOG1.MVRP to about 75% of max duration
devmem2 0x20101010 w 0x00BFFFF4
# Set WATCHDOG1.TRIGGER to its max value
devmem2 0x20101014 w 0x00000FFF
# Set WATCHDOG1.RUNTIME to max duration (about 30 seconds)
devmem2 0x2010100C w 0x00FFFFF0
# Set WATCHDOG1.CTRL to (4:forbidden_enabled + 1:wdog_irq_enabled + 0:mvrp_irq_enabled)
devmem2 0x20101004 w 0x13
# Write into WATCHDOG1.REFRESH to start watchdog 1
devmem2 0x20101000 w 0xDEADC0DE

# -- Check watchdog 1 state --
# Check WATCHDOG1 counter value
devmem2 0x20101000 w
# Check WATCHDOG1 status
devmem2 0x20101008 w
```

Here is below the HSS log, taken from the first watchdog timeout trigger event.
Notice the lines 'Watchdog has triggered - 10' appearing 10 times, roughly every 30 seconds.
The system finally gets restarted at the [419.13129] time mark, after 9 failed attempts.

Thanks for your inputs.

```
[160.253111] Watchdog has triggered - 10
[160.258718] Boot image passed CRC
[160.263683] Boot image set name: "PolarFire-SoC-HSS::U-Boot"
[160.271009] Boot image passed CRC
[160.275974] Boot image set name: "PolarFire-SoC-HSS::U-Boot"
[160.283300] Boot image passed CRC
[160.288265] Boot image set name: "PolarFire-SoC-HSS::U-Boot"
[160.295590] Boot image passed CRC
[160.300555] Boot image set name: "PolarFire-SoC-HSS::U-Boot"
[160.307812] boot_service(u54_1) :: [Init] -> [SetupPMP]
[160.314877] boot_service(u54_2) :: [Init] -> [SetupPMP]
[160.321943] boot_service(u54_3) :: [Init] -> [SetupPMP]
[160.329008] boot_service(u54_4) :: [Init] -> [SetupPMP]
[160.336074] boot_service(u54_1)::Registering domain "u-boot.bin" (hart mask 0x1e)
[160.345335] boot_service(u54_1) :: [SetupPMP] -> [SetupPMPComplete]
[160.353547] boot_service(u54_2) :: [SetupPMP] -> [SetupPMPComplete]
[160.361758] boot_service(u54_3) :: [SetupPMP] -> [SetupPMPComplete]
[160.369969] boot_service(u54_4) :: [SetupPMP] -> [SetupPMPComplete]
[160.378181] u54 State Change:  [Booting] [Booting] [Booting] [Booting]
[160.387729] boot_service(u54_1) :: [SetupPMPComplete] -> [ZeroInit]
[160.395940] boot_service(u54_2) :: [SetupPMPComplete] -> [ZeroInit]
[160.404151] boot_service(u54_3) :: [SetupPMPComplete] -> [ZeroInit]
[160.412362] boot_service(u54_4) :: [SetupPMPComplete] -> [ZeroInit]
[160.420574] boot_service(u54_1) :: [ZeroInit] -> [Download]
[160.428021] boot_service(u54_2) :: [ZeroInit] -> [Download]
[160.435469] boot_service(u54_3) :: [ZeroInit] -> [Download]
[160.442916] boot_service(u54_4) :: [ZeroInit] -> [Download]
[... 507 lines cut ...]
[7.642202] boot_service(u54_3) :: [SetupPMPComplete] -> [ZeroInit]
[7.650222] boot_service(u54_4) :: [SetupPMPComplete] -> [ZeroInit]
[7.658243] boot_service(u54_1) :: [ZeroInit] -> [Download]
[7.665499] boot_service(u54_2) :: [ZeroInit] -> [Download]
[7.672756] boot_service(u54_3) :: [ZeroInit] -> [Download]
[7.680012] boot_service(u54_4) :: [ZeroInit] -> [Download]
[7.687269] boot_service(u54_1)::Processing boot image: "u-boot.bin"
[7.695098] boot_service(u54_2) :: [Download] -> [Idle]
[7.701973] boot_service(u54_3) :: [Download] -> [Idle]
[7.708847] boot_service(u54_4) :: [Download] -> [Idle]
[7.764382] boot_service(u54_1) :: [Download] -> [OpenSBIInit]
[7.771925] boot_service(u54_1)::u54_2:sbi_init 0x1000200000
[7.778990] boot_service(u54_1)::u54_3:sbi_init 0x1000200000
[7.786056] boot_service(u54_1)::u54_4:sbi_init 0x1000200000
[7.793121] boot_service(u54_1) :: [OpenSBIInit] -> [Wait]
[7.800282] boot_service(u54_1)::u54_1:sbi_init 0x1000200000
[7.808684] boot_service(u54_1)::Checking for IPI ACKs: - -
[7.863967] boot_service(u54_1)::Checking for IPI ACKs: ACK/IDLE ACK
[7.892516] boot_service(u54_1) :: [Wait] -> [Idle]
[7.910275] u54 State Change:  [SBIHartInit] [Booting] [Booting] [Booting]
[7.944839] loop 2706 took 64506354 ticks (max 64506354 ticks)
[8.45570] u54 State Change:  [Running] [SBIWaitForColdboot] [SBIWaitForColdboot] [SBIWaitForColdboot]
[12.823494] u54 State Change:  [Running] [Running] [Running] [Running]
[37.478383] loop 2500000 took 4732 ticks (max 64506354 ticks)
```

griffini, 2022-11-15:
Hi @nearly-big-endian, thanks for reporting this.

We'll attempt to reproduce this and then diagnose what is going on. I suspect a small race condition in the HSS in how the individual harts are restarting when the watchdog trips.

griffini, 2022-11-21:
Hi @nearly-big-endian , we have a patch being reviewed to address this issue for you.

nearly-big-endian, 2022-11-21:
> Hi @nearly-big-endian , we have a patch being reviewed to address this issue for you.

This is great news, thanks. I am looking forward to trying this fix.

griffini, 2022-11-24:
Hi @nearly-big-endian 

For now the following workaround patch will restore the behavior you want. A clean fix for this (which also supports AMP correctly) will be added into the next official HSS release, but we didn't want to hold you up in the interim.

```
diff --git a/services/wdog/wdog_service.c b/services/wdog/wdog_service.c
index cafff21..98327bc 100644
--- a/services/wdog/wdog_service.c
+++ b/services/wdog/wdog_service.c
@@ -153,6 +153,9 @@ static void wdog_monitoring_handler(struct StateMachine * const pMyMachine)
     status &= hartBitmask.uint;
 
     if (status) {
+#if IS_ENABLED(CONFIG_ALLOW_COLDREBOOT_ALWAYS)
+        HSS_Wdog_Reboot(HSS_HART_ALL);
+#else
         // watchdog timer has triggered for a monitored hart..
         mHSS_DEBUG_PRINTF(LOG_ERROR, "Watchdog has triggered - %02x\n", status);
 
@@ -176,6 +179,7 @@ static void wdog_monitoring_handler(struct StateMachine * const pMyMachine)
             HSS_Boot_RestartCore(HSS_HART_U54_4);
             wdogInitTime[HSS_HART_U54_4] = HSS_GetTime();
         }
+#endif
 #endif
     }
 }

```

nearly-big-endian, 2022-11-24:
Hi,

I am very pleased to report that we have tested this patch and that it fixes the issue we were having perfectly.

Thanks very much for the quick response (and correction).

closed 2022-11-24 by nearly-big-endian