polarfire-soc/hart-software-services #71: Illegel instruction error for U54 harts after booting up

sezgin-karapinar, 2024-07-04:
Hello everyone,

I used hss payload generator. It only includes elf file of mpfs gpio interrup example project built with DDR-Release option as payload. When I started the system, HSS prints out messages given below. After that, states of U54 harts changes from running to fatal. How can I correct this ?

[4.914845] Address to execute is 80200000
sbi_trap_error: hart1: illegal instruction handler failed (error -2)
sbi_trap_error: hart1: mcause=0x0000000000000002 mtval=0x0000000000000000
sbi_trap_error: hart1: mepc=0x00000000801921c2 mstatus=0x0000000a00001800
sbi_trap_error: hart1: ra=0x000000000a003362 sp=0x000000000a04be08
sbi_trap_error: hart1: gp=0x0000000000000000 tp=0x000000000a027620
sbi_trap_error: hart1: s0=0x0000000a00001800 s1=0x0000000000000003
sbi_trap_error: hart1: a0=0x0000000000000001 a1=0x0000000000000000
sbi_trap_error: hart1: a2=0x0000000000000007 a3=0x0000000000000019
sbi_trap_error: hart1: a4=0x0000000000000001 a5=0x0000000000000001
sbi_trap_error: hart1: a6=0x0000000000000010 a7=0x0000000000000010
sbi_trap_error: hart1: s2=0x0000000080200000 s3=0x0000000000000000
sbi_trap_error: hart1: s4=0x0000000000000001 s5=0x000000000a017e6a
sbi_trap_error: hart1: s6=0x000000000a02e2b8 s7=0x000000000a01dbb8
sbi_trap_error: hart1: s8=0x0000000000004000 s9=0x0000000000000000
sbi_trap_error: hart1: s10=0x0000000000000000 s11=0x0000000000000000
sbi_trap_error: hart1: t0=0x0000000a00001880 t1=0x000000000a009d3c
sbi_trap_error: hart1: t2=0x0000000000000001 t3=0x0000000000000000
sbi_trap_error: hart1: t4=0x00000000801b6bdc t5=0x0000000000000000
sbi_trap_error: hart1: t6=0x0000000000000000

griffini, 2024-09-26:
The first thing I would suggest to do is make sure that there is a valid instruction at 0x801921c2? (the MEPC address).

You could connect with a debugger, and view that address to be sure.  If not, then either something is corrupting/overwriting memory, or your payload generator configuration may be wrong.

griffini, 2024-12-20:
Hi @sezgin-karapinar , did you make an progress with your issue?

closed 2025-06-04 by griffini