polarfire-soc/hart-software-services #73: One CAN channel does not work properly when using the 2023.09 version of HSS

ZGDIGE, 2024-09-19:
Using solfconsole for DEBUG testing, both CAN channels are working fine. But after using HSS, I found that CAN0 channel is not working at all, the CAN1 channel can work nomally.

griffini, 2024-09-26:
Hi @ZGDIGE 

Can you provde some more details on what exactly it is you are doing? Specifically:

- are you using a custom design? If so, did you update the XML file for your design?
- what is the value of LIBERO_SETTING_CONFIGURED_PERIPHERALS ?

griffini, 2025-06-04:
@ZGDIGE thank you for reporting this issue and providing details about your setup. Since there hasn't been an update in several months and we are unable to investigate further without additional information, we’ll close this issue for now.

If you’re still experiencing this problem with CAN0 functionality after using HSS, please feel free to:
* reopen this issue with the requested details (e.g., XML configuration, LIBERO_SETTING_CONFIGURED_PERIPHERALS value, or a minimal reproducible example), or
* Create a new issue if there have been updates to your environment or new insights.

We appreciate your contribution to improving this project and are happy to revisit this if you’re able to provide more context.

closed 2025-06-04 by griffini