polarfire-soc/hart-software-services #74: FPGENPROG environment variable needs to point to the binary directly

cmcahoon01, 2024-09-24:
I am following the README instructions for building on Linux.
I have gotten to the point of running `make` and encounter this error
```
12:02:19 INFO  - mpfsBootmodeProgrammer v3.7 started.
12:02:19 INFO  - "/root/br/hart-software-services/build/bootmode1" is the output folder and the previous contents of this folder will be deleted.
12:02:19 INFO  - Selected boot mode "1 - non-secure boot from eNVM" and working in directory "/root/br/hart-software-services/build".

12:02:19 ERROR - The FPGENPROG environment variable needs to point to the binary directly (including the "fpgenprog" binary file).

cp: cannot stat 'build/bootmode1/hss-envm-wrapper-bm1-p0.hex': No such file or directory
   text    data     bss     dec     hex filename
 102052     416  149632  252100   3d8c4 build/hss-envm-wrapper.elf
```

The error implies my FPGENPROG variable is not set up properly.
Here is an `ls` in my installed directory, and the $FPGENPROG variable matches the path.
[image]
Is the problem with my path, or that I'm using root? 
Thanks

griffini, 2024-09-26:
The path looks good. In my case, for example, it is set to `/opt/microchip/Libero_SoC_v2024.1/Libero/bin64/fpgenprog`.

Can you try as a user other than root?

cmcahoon01, 2024-09-26:
I tried as a non root user, but did not have any different results. 
I'm aware that this won't be easily diagnosed like this, but I appreciate you trying to help.

closed 2024-09-26 by cmcahoon01