polarfire-soc/hart-software-services #75: Polarfire Discovery board booting issue with SD card

Embedded-linux, 2024-12-02:
[discussion-forum]: https://github.com/polarfire-soc/polarfire-soc-documentation/discussions

- [ ] Provide a minimal code snippet example that reproduces the bug where appropriate.
I am using, pre-built  images of microchip

- [ ] Provide screenshots where appropriate
Will add the SD card issue with the board.
- [ ] Does this occur all of the time or is it intermittent?
It occurs all the time
- [ ] Have you tried on another board?
No, I have  one board with me.

This is Satish, I recently purchased board MPFS Discovery kit
Have programmed Linux on eNVM using reference file
https://github.com/polarfire-soc/polarfire-soc-discovery-kit-reference-design/releases/tag/2024.04

have copied boot images on SD card as below
(pre built images) created 2 partitions 

1 st part: boot.scr      fitImage      uboot.env
2nd part: core-image-minimal-xxx. wic

part 1: W95 FAT system (0b)
part 2: Linux (83)
[image]
[image]

created file system of type:

mkfs.vfat /dev/sda1 -n boot
mkfs.ext4 /dev/sda2 -L root

but SD card is not identified by the board.
Have tried multiple options in file systems,still it is keep failing.

I am using pre built images provided by Microchip
where it is going wrong?
can you please suggest & provide the reference pages for Linux booting on Discovery kit.

I have tried earlier, SD card,  boot is successful with kernel &  failed with kernel panic, so did the format of SD card & created new partitions, it is unable to identify the SD card

Observations:  Successful case: created 2 partitions 
copied 3 files in boot folder: boot.scr      fitImage      uboot.env
second partition is empty: **no file system**
so it is failing because there is no file system

Failure case :  1 st partition : Copied 3 images to boot folder of type FAT 
                        2 nd partition : copied core-image-minimal-xxx.wic   of type Linux 
it is keep failing, let me add the successful case, screen shots

Thanks in advance 

Regards
Satish G

Embedded-linux, 2024-12-02:
[image]
[image]
[image]
[image]

griffini, 2024-12-02:
Hi @Embedded-linux ,

Thanks for reporting your issue. I'm not sure if I fully understand what you did -- but if you have *manually* created partitions etc., then this is a mistake.

The WIC image must be written directly to the SDCard in a raw sector-by-sector fashion. This will ensure the correct partition table is created.   You may find the instructions at https://mi-v-ecosystem.github.io/redirects/boards-mpfs-generic-updating-mpfs-kit useful for doing this.

If you have followed these instructions and still have issues, please let us know exactly what steps you followed.

Embedded-linux, 2024-12-05:
Hi @griffini 

I am getting Issue while copying .bmap file to SD card.
Obsrvation:  bmap file: It is file extension of type .xml, uses to copy the images in faster way.
Reference: https://wayneoutthere.com/2020/06/19/how-to-bmap-tool-thing-ubuntu/

bmap file is used to copy "img.xz" file to SD card "/mount/media/satish/root"  
command: sudo bmaptool copy --bmap ~/Downloads/xxxxxx.bmap ~/Downloads/**ubuntu-rootfile.img.xz** /dev/mmcblk0

In the above command .bmap is used as support file to copy the .img.xz file to SD card. 
Now from Micro chip polarfire reference boards, we have only .bmap files, have n't seen img.xz files
reference: https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases?page=1

so which image needs to copy to SD card ?

Please support me to understand the bmap usage, If I am wrong

Thanks & Regards
Satish G

Embedded-linux, 2024-12-11:
Hi @griffini 

than you for the response
I find out way to copy the .wic images to SD card using below command
**sudo dd bs=4M if=core-image-minimal-dev-mpfs-disco-kit-20240426091646.rootfs.wic of=/dev/sda status=progress && sync**

copies successfully to SD card & power on the board.

It is identifying SD card but it is failing in getting MMC 1 partition.

[image]

Adding the screen shots for the same.
Can you suggest me , what is getting wrong

Thanks in advance.

Thanks & Regards
Satish G

Embedded-linux, 2024-12-11:
[image]

Embedded-linux, 2024-12-11:
[image]

Embedded-linux, 2024-12-11:
Adding screen shot of Issue , am getting with MMC card when I copied the Image with bmaptool 
**sudo bmaptool copy --bmap core-image-minimal-dev-mpfs-disco-kit-20240703233429.rootfs.wic.bmap core-image-minimal-dev-mpfs-disco-kit-20240703233429.rootfs.wic /dev/sda1**

Thanks & Regards
Satish G

griffini, 2024-12-11:
> Adding screen shot of Issue , am getting with MMC card when I copied the Image with bmaptool **sudo bmaptool copy --bmap core-image-minimal-dev-mpfs-disco-kit-20240703233429.rootfs.wic.bmap core-image-minimal-dev-mpfs-disco-kit-20240703233429.rootfs.wic /dev/sda1**
> 
> Thanks & Regards Satish G

Hi @Embedded-linux , I can't seem to see your screenshot?   WHere did you post it?

Can you describe what the problem you are seeing is?  
Can you also provide the output from running `lsblk` on your system?

I suspect `/dev/sda1` is a *partition* of some kind on your platform, and not a full drive? You need to do the copy to the raw drive so that the partition tables are correctly set.  Also, is `/dev/sda` the correct device? I can confirm with the output from `lsblk`.

Embedded-linux, 2024-12-11:
Hi @griffini 

ok, let me add my booting steps in sequence to avoid confusion

1. Copy the .bmap & .bmap.bz images from polar fire below url
    https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/releases/tag/v2024.06
    downloaded images: core-image-minimal-dev-mpfs-disco-kit-20240703233429.rootfs.wic.bmap   
                                     core-image-minimal-dev-mpfs-disco-kit-20240703233429.rootfs.wic.gz
2. uncompress * .wic.gz image as .wic image using gzip -d " xxxx.gz"
3. Copy the images to SD card using below command
    sudo bmaptool copy --bmap core-image-minimal-dev-mpfs-disco-kit-20240703233429.rootfs.wic.bmap core-image-minimal-dev-mpfs-disco-kit-20240703233429.rootfs.wic /dev/sda

4. output of **lsblk -o name,label,size,fstype,fsver /dev/sda**

satish@IdeaPad:~/Microchip/Linux-image$ lsblk -o name,label,size,fstype,fsver /dev/sda
**NAME   LABEL   SIZE FSTYPE FSVER
sda            7.4G        
├─sda1 boot     64M vfat   FAT16
├─sda2           8M        
└─sda3 root  698.7M ext4   1.0**

Question: why it is creating 3 partitions, as 2 partitions are good enough as boot & root partitions ?

5. Un mount SD card & insert  the SD card in the target board
getting the issue as below

Thanks & Regards
Satish G

Embedded-linux, 2024-12-11:
[image]

Embedded-linux, 2024-12-11:
[image]

Embedded-linux, 2024-12-11:
[image]

Embedded-linux, 2024-12-11:
[image]

Embedded-linux, 2024-12-11:
[image]

Embedded-linux, 2024-12-12:
Hi @griffini 

While copying the xxx.wic image to SD card using bmap tool
it is creating 3 partitions, in general, it requires 2 partitions, 
what is use of 3 rd partition & 
hope missing part is,  file type of partition
By default, boot image will work with FAT file system, here FAT32, FAT16 
& root folder is of type **"ext4"**

In my case, after copying the Images into SD card through bmap tool
can see the default File system types as below
adding the screen shot
[image]

suggest me to find out, file types of the Partitions 
existing successful boot SD card is having which file system types?

Hope that will resolve the issue

Thanks & Regards
Satish G

griffini, 2024-12-12:
Hi @Embedded-Linux,

Three partitions are correct. One for root filesystem, one for /boot, and a BIOS boot partition containing the bootable payload next stage that the HSS looks for (in this case, U-Boot).

If you have copied the partitions using bmap correctly to the raw device (as your command line above suggests), you should not need to concern yourself with the partition types. They will be correctly created by the bmap copy step.

Can you tell me what SD card you are using (manufacturer, and card model)?

Embedded-linux, 2024-12-12:
Hi @griffini ,

thank you for the response
I am using 8GB SD card & It is of type  "Sandisk ultra" model from Sandisk.

Thanks & regards
Satish G

Embedded-linux, 2024-12-17:
Hi @griffini ,

Now kernel & rootfs are coming successfully
Earlier problem:  existing SD card is not supporting "polar fire" file format versions 
Solution: Have gone through multiple blogs & connected with polar fire community, have changed SD card of type new.
              
Next step: Port Zephyr RTOS on Polar fire.

Thank you for continuous support

[image]
[image]
[image]
[image]
   
Thanks & Regards
Satish G

griffini, 2024-12-17:
@Embedded-linux ,

Fantastic to hear it is working for you now, thanks for letting me know!

For Zephyr,  you may find https://github.com/polarfire-soc/zephyr and https://github.com/polarfire-soc/polarfire-soc-zephyr-applications useful.

Embedded-linux, 2024-12-19:
Hi @griffini ,

I have gone through the above Zephyr related URLs,
Thank you for the support.
 
Requirement : Need to Port Zephyr OS , instead of Linux & will compile the custom python libraries on top of Zephyr.
Based on above requirement, do we have any reference page,  **"Compilation procedure & port the Zephyr on Polar fire Discovery kit."**

It will be great help & thanks in advance

Thanks & Regards
Satish G