polarfire-soc/hart-software-services #76: UART error occurs when HSS is initialized

ZGDIGE, 2024-12-19:
Hello,
The uart configuration in HSS is the same as that in putty. When running to the uart_getchar function, a Parity error status appears, resulting in failure to input ymodem.

[image]

[image]
[image]

griffini, 2024-12-19:
Hi @ZGDIGE ,

This is very strange. And I notice you're getting a UART error even *before* you start YMODEM.

A parity error occurs when the expected calculated parity by the receiver doesn't match the parity bit sent by the transitter. Since you presumably are using 115200 8N1, then parity is disabled and no bit is transmitted, how are you certain that it is a parity error?

I don't seem to see this with my Icicle kit.

With 8N1 I would expect that a parity error is unlikely, and the issue is more likely  to be a framing error or bit-level corruption? Could you be seeing something causing clock drift or incorrect interpretation of start/stop bits? For example, electrical noise or crosstalk or other signal integrity issues? Or a baud rate mismatch (clock inaccuracies or jitter?)? Or improper level shifting/buffering?

ZGDIGE, 2024-12-20:
> Hi @ZGDIGE ,
>
> This is very strange. And I notice you're getting a UART error even _before_ you start YMODEM.
>
> A parity error occurs when the expected calculated parity by the receiver doesn't match [...]

Thank you for your answer. I found that it was a problem with the hardware link. We used a moxa device to convert UART to RS422. After changing the conversion device, there was no problem.

griffini, 2024-12-20:
Hi @ZGDIGE, Great to hear you have your issue resolved, and thanks for letting me know!

closed 2024-12-20 by griffini