polarfire-soc/hart-software-services #81: HSS build failed with "fatal error: inttypes.h: No such file or directory"

anthony-d-truong, 2025-03-24:
Ubuntu Linux 22.04
Master branch.
SoftConsole v2022.2-RISC-V-747
Libero_SoC v2024.2

make BOARD=mpfs-icicle-kit-es  yields:
INFO: Linux detected
INFO: Linux Desktop detected
mpfs-icicle-kit-es selected
INFO: NOTICE: enabling -flto (which means stack protection is disabled)
INFO: Expected riscv64-unknown-elf-gcc version 8.3.0 but found 10.2.0
INFO: Expected mpfsBootmodeProgrammer.jar version v3.6 but found 
INFO: This version of the HSS relies on SoftConsole v2021.3 or later
INFO: Linux detected
INFO: Linux Desktop detected
mpfs-icicle-kit-es selected
INFO: NOTICE: enabling -flto (which means stack protection is disabled)
INFO: Expected riscv64-unknown-elf-gcc version 8.3.0 but found 10.2.0
INFO: Expected mpfsBootmodeProgrammer.jar version v3.6 but found 
INFO: This version of the HSS relies on SoftConsole v2021.3 or later
 MPFSCFGGEN    boards/mpfs-icicle-kit-es/soc_fpga_design/xml/ICICLE_MSS_mss_cfg.xml
Input XML file: /home/anthony.dominic/projects/polarfire/hart-software-services/boards/mpfs-icicle-kit-es/soc_fpga_design/xml/ICICLE_MSS_mss_cfg.xml
Hardware configuration header files created in directory: /home/anthony.dominic/projects/polarfire/hart-software-services/build/boards/mpfs-icicle-kit-es/fpga_design_config
 GENCONFIG
 CC        build/application/hart0/hss_state_machine.o
In file included from application/hart0/hss_state_machine.c:22:
./include/hss_debug.h:44:14: fatal error: inttypes.h: No such file or directory
   44 | #    include "inttypes.h"
      |              ^~~~~~~~~~~~
compilation terminated.
make: *** [application/rules.mk:184: build/application/hart0/hss_state_machine.o] Error 1

rockoguido, 2025-03-25:
> EXTERNAL EMAIL: Do not click links or open attachments unless you know the content is safe

Hi..

Assuming you are using the latest HSS..   version 2024.09.

Be sure to include the following steps before your build..

export SC_INSTALL_DIR=/home/user/Microchip/SoftConsole-v2022.2

For building on Linux from the command line, configure the path appropriately. For example, the following will add the SoftConsole provided Python and RISC-V compiler toolchain to the path:

    $ export PATH=$PATH:$SC_INSTALL_DIR/python3/bin:$SC_INSTALL_DIR/riscv-unknown-elf-gcc/bin

These steps come from the README at the root of the HSS distro.

After that your make sure proceed without an issue.    assuming you already have a .config in the root directory.

make BOARD=mpfs-icicle-kit-es

I just did a quick test to verify this….

Regards,

Dave Campanella
Principal Embedded Solutions Engineer / FAE
Microchip Technology Inc.
165 Technology Dr
Irvine, CA 92618
+1 925 989 0569 (mobile)
+1 760 304 0172 (office)
***@***.******@***.***>

For any Technical related concerns, kindly you can create web ticket at http://www.microchip.com/support
For Technical Training please go to Microchip University at https://secure.microchip.com/mu

There are several paths to take for Microchip support:
        * Self-paced learning – see the Developer Help Site<http://microchipdeveloper.com/>
        * Knowledge Base - here<http://www.microchip.com/support/hottopics.aspx>
        * Community Forums - help<http://www.microchip.com/forums/> with a lot of information from experts.

anthony-d-truong, 2025-03-25:
Thank you, Dave.
Same problem.  Please see attached.

[image]

anthony-d-truong, 2025-03-25:
Hello Dave,
I do have the latest versions of those packages in my Ubuntu 22.04 laptop.  Same problem.  When I tried bulding HSS with SoftConsole v2022.2, I got the same build error.

rockoguido, 2025-03-25:
Hi Anthony,

Well to be honest if your host environment is ok then your HSS distro is most likely flawed in some way.

What I would do is remove what you have for HSS.    On a windows machine just download the archive and try to build from SoftConsole on Windows for starters.   Just  a way to verify the distribution, low hanging fruit and then migrate to Linux and do the same.   Just download the archive and build within SoftConsole.

Other option could be a flawed SoftConsole download?    Can you build any of our bare metal examples with SoftConsole?
See https://github.com/polarfire-soc/polarfire-soc-bare-metal-examples

Regards,

Dave Campanella
Principal Embedded Solutions Engineer / FAE
Microchip Technology Inc.
165 Technology Dr
Irvine, CA 92618
+1 925 989 0569 (mobile)
+1 760 304 0172 (office)
***@***.******@***.***>

For any Technical related concerns, kindly you can create web ticket at http://www.microchip.com/support
For Technical Training please go to Microchip University at https://secure.microchip.com/mu

There are several paths to take for Microchip support:
        * Self-paced learning – see the Developer Help Site<http://microchipdeveloper.com/>
        * Knowledge Base - here<http://www.microchip.com/support/hottopics.aspx>
        * Community Forums - help<http://www.microchip.com/forums/> with a lot of information from experts.