polarfire-soc/hart-software-services #85: Rebuilding the DTB is not Supported when building using SoftConsole v2022.2-RISC-V-747

720adams, 2025-06-01:
I'm trying to build the latest version of the hart-software-services repo with the latest SoftConsole tools v2022.2-RISC-V-747 on Rocky 8 Linux.  A lot of the repo builds and then I run into this issue:

13:13:27 **** Build of configuration build for project hart-software-services ****
make all 
INFO: Linux detected
INFO: Linux Desktop detected
INFO: BOARD not specified, defaulting to mpfs-icicle-kit-es
mpfs-icicle-kit-es selected
INFO: NOTICE: enabling -flto (which means stack protection is disabled)
INFO: Expected mpfsBootmodeProgrammer.jar version v3.6 but found 
INFO: This version of the HSS relies on SoftConsole v2021.3 or later
INFO: Linux detected
INFO: Linux Desktop detected
INFO: BOARD not specified, defaulting to mpfs-icicle-kit-es
mpfs-icicle-kit-es selected
INFO: NOTICE: enabling -flto (which means stack protection is disabled)
INFO: Expected mpfsBootmodeProgrammer.jar version v3.6 but found 
INFO: This version of the HSS relies on SoftConsole v2021.3 or later
 GENCONFIG
 CC        build/application/hart0/hss_state_machine.o
 CC        build/application/hart0/hss_clock.o
 CC        build/application/hart0/hss_registry.o
 CC        build/baremetal/drivers/mss/mss_usb/mss_usb_common_cif.o
 CC        build/baremetal/drivers/mss/mss_usb/mss_usb_device.o
 CC        build/baremetal/drivers/mss/mss_usb/mss_usb_device_cif.o
 CC        build/baremetal/drivers/mss/mss_usb/mss_usb_device_msd.o
 CC        build/baremetal/polarfire-soc-bare-metal-library/src/platform/hal/hal_irq.o
 CC        build/baremetal/polarfire-soc-bare-metal-library/src/platform/drivers/mss/mss_gpio/mss_gpio.o
 CC        build/services/beu/beu_service.o
 CC        build/services/boot/hss_boot_service.o
 CC        build/services/boot/hss_boot_pmp.o
 CC        build/services/boot/gpt.o
 CC        build/services/ddr/ddr_service.o
 CC        build/services/ddr/ddr_api.o
 CC        build/services/goto/goto_service.o
 CC        build/services/healthmon/healthmon_service.o
 CC        build/services/healthmon/healthmon_monitors_weak.o
 CC        build/services/ipi_poll/ipi_poll_service.o
 CC        build/services/opensbi/opensbi_service.o
 CC        build/thirdparty/opensbi/lib/utils/irqchip/plic.o
 CC        build/thirdparty/opensbi/lib/utils/libfdt/fdt.o
 CC        build/thirdparty/opensbi/lib/utils/libfdt/fdt_ro.o
 CC        build/thirdparty/opensbi/lib/utils/libfdt/fdt_rw.o
 CC        build/thirdparty/opensbi/lib/utils/libfdt/fdt_wip.o
 CC        build/thirdparty/opensbi/lib/utils/libfdt/fdt_addresses.o
 CC        build/thirdparty/opensbi/lib/utils/fdt/fdt_fixup.o
 CC        build/thirdparty/opensbi/lib/utils/fdt/fdt_helper.o
 CC        build/thirdparty/opensbi/lib/sbi/riscv_asm.o
 CC        build/thirdparty/opensbi/lib/sbi/riscv_atomic.o
 CC        build/thirdparty/opensbi/lib/sbi/riscv_locks.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_bitmap.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_bitops.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_console.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_domain.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_ecall.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_emulate_csr.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_fifo.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_hart.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_hsm.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_illegal_insn.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_init.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_ipi.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_irqchip.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_math.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_misaligned_ldst.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_platform.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_pmu.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_scratch.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_string.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_system.o
[... 73 lines cut ...]
 CC        build/modules/debug/hss_perfctr.o
 CC        build/modules/misc/assert.o
 CC        build/modules/misc/csr_helper.o
 CC        build/modules/misc/c_stubs.o
 CC        build/modules/misc/hss_crc16.o
 CC        build/modules/misc/hss_crc32.o
 CC        build/modules/misc/hss_memcpy_via_pdma.o
 CC        build/modules/misc/hss_progress.o
 CC        build/modules/misc/device_serial_number.o
 CC        build/modules/misc/design_version_info.o
 CC        build/modules/misc/uart_helper.o
 CC        build/modules/misc/hss_trigger.o
 CC        build/modules/misc/hss_memtest.o
 CC        build/application/crt.o
 CC        build/baremetal/polarfire-soc-bare-metal-library/src/platform/mpfs_hal/startup_gcc/mss_utils.o
 CC        build/thirdparty/opensbi/lib/sbi/riscv_hardfp.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_expected_trap.o
 CC        build/thirdparty/opensbi/lib/sbi/sbi_hfence.o
 CC        build/services/opensbi/opensbi_ecall_exts.o
services/opensbi/Makefile:143: *** Error:: Rebuilding the DTB is not supported... Please rebuild manually.  Stop.
"make all" terminated with exit code 2. Build might be incomplete.

13:14:14 Build Failed. 1 errors, 0 warnings. (took 46s.773ms)

Thoughts of what to do?

griffini, 2025-06-01:
SoftConsole does not include the device tree compiler tool (dtc), and for some reason your .dts file is newer than your .dtb.

You can either rebuild the dtb manually (using dtc):

```
$ dtc -I dts -O dtb -o boards/mpfs-icicle-kit/mpfs_minimal_uboot.dtb boards/mpfs-icicle-kit/mpfs_minimal_uboot.dts
```

 or just use the touch command to update the timestamp on the dtb and your build should work:

```
$ touch boards/mpfs-icicle-kit/mpfs_minimal_uboot.dtb
```

720adams, 2025-06-03:
griffini:  Thanks so much for the response.  Unfortunately neither suggestion worked for me.  I can't find a "dtc" compiler in any of the tools from Microchip that I'v installed.  Where might I find it?

I also tried touching hart-software-services/boards/mpfs-icicle-kit-es/mpfs_minimal_uboot.dtb to make it's date more recent than hart-software-services/boards/mpfs-icicle-kit-es/mpfs_minimal_uboot.dts.  That didn't work either.  I still get the same failure when trying to build.

rockoguido, 2025-06-03:
> EXTERNAL EMAIL: Do not click links or open attachments unless you know the content is safe

I’ll just add a quick comment.    You won’t find the device tree compiler (DTC) as part of Microchip’s tools.   It’s part of your Linux host tools

Regards,

Dave Campanella
Principal Embedded Solutions Engineer / FAE
Microchip Technology Inc.
165 Technology Dr
Irvine, CA 92618
+1 925 989 0569 (mobile)
+1 760 304 0172 (office)
***@***.******@***.***>

***@***.***

griffini, 2025-06-04:
Yes, `dtc` comes with your Linux host system, it is not a Microchip specific tool.  For Rocky Linux 9, you can install it using `dnf install dtc`. For Rocky Linux 8, you can get it using `dnf --enablerepo=powertools install dtc`.

The error you are encountering is detected in `services/opensbi/Makefile`:

```make
$(CONFIG_DEFAULT_DEVICE_TREE_UNQUOTED): $(CONFIG_DEFAULT_DEVICE_TREE_UNQUOTED:%.dtb=%.dts)
        $(error Error:: Rebuilding the DTB is not supported... Please rebuild manually)
```

It is due to the DTB being newer than the DTS file, so touching it will fix the issue. If you are still getting the issue, perhaps you touched the wrong file?  Can you ls both files and compare the timestamps?

Also, perhaps look at your `.config` file and ensure that `CONFIG_DEFAULT_DEVICE_TREE` is set appropriately (and that you are touching the corresponding dts file)?

720adams, 2025-06-05:
griffini thanks for the continued responses:

I checked and the only dtb file in the whole directory was ./hart-software-services/boards/mpfs-icicle-kit-es/mpfs_minimal_uboot.dtb

I checked the timestamp, and it was newer than the .dts file of the same name

Neither of those fixed the problem.  I was able to load the dtc tool under Rocky 8 Linux using the command you suggested.  I even rebuilt the dtb file using the command you formerly suggested, but I still get the same results when I run the build with Softconsole from a default "git clone" image imported with Softconcole.  Finally, I deleted those lines from the Makefile and I reload the project.  Then the tool builds the hss-envm-wrapper.elf and hss-l2scratch.elf.

I was doing all this to build hss.bin so I can get QEMU running for the icicle board using the command:

qemu-system-riscv64 -M microchip-icicle-kit -smp 5 \
   -bios path/to/hss.bin -sd path/to/sdcard.img \
   -nic user,model=cadence_gem \
   -nic tap,ifname=tap,model=cadence_gem,script=no \
   -display none -serial stdio \
   -chardev socket,id=serial1,path=serial1.sock,server,wait \
   -serial chardev:serial1

This command comes from the website:  https://wiki.qemu.org/Documentation/Platforms/RISCV

I'm obviously still missing some pieces to the puzzle.  Any further guidance?

lranders, 2025-11-20:
@720adams:
You don't actually need `dtc`...

All you need to do is `touch -r <dts file> <dtb_file>` and your problem goes away.