polarfire-soc/hart-software-services #89: Understanding HSS and Adding Custom Boot Support for Automatic Boot Switch Between Main and Redundant QSPI

ankrtlpixxel, 2025-09-25:
Hello,

I am working on custom boot support logic for a PolarFire SoC where Linux boot should automatically happen from the main QSPI memory, and if that fails, it should try booting from a redundant QSPI. I am exploring ways to map two different QSPI devices to the HSS (Hart Software Services) configuration or implement switching between them.

**Questions**:

- Custom Bootloader Implementation : I am planning to develop a custom bootloader that drives chip select (CS) pins through GPIO and performs switching based on certain checks. Can the boot code running in eNVM (embedded Non-Volatile Memory) be modified to support this functionality?
- ROM Boot Code Configuration : How can I ensure that the ROM boot code in eNVM only loads from a selected address range in the QSPI where I have placed different components (device tree, kernel image, root file system, and bootloader)? Where is this boot address/offset configured?
- FPGA IO Control from eNVM : Once the FPGA fabric is configured and running, can the eNVM code drive/read the FPGA IO pins in a baremetal flow, similar to how the selection between eMMC and SD card is implemented in Icicle development kits?

**Context**:
- I need to understand the boot flow customization options available in PolarFire SoC and how to implement redundant boot sources for increased reliability.

griffini, 2025-09-26:
The HSS boot logic in the PolarFire SoC is designed to be extensible via the `struct HSS_Storage` abstraction. You can add custom storage providers for main and redundant QSPI, each with its own chip select logic and address range. The boot code in eNVM can be modified to support this, and the boot process will automatically try each storage provider in order, enabling fallback and redundancy. 

You can also implement custom GPIO or FPGA IO logic in your storage provider functions to control chip selects or perform health checks. 

The address/offset for booting from QSPI is configurable via macros and your storage provider implementation.

* **Custom Bootloader Implementation : I am planning to develop a custom bootloader that drives chip select (CS) pins through GPIO and performs switching based on certain checks. Can the boot code running in eNVM (embedded Non-Volatile Memory) be modified to support this functionality?**

The HSS boot logic is built around the concept of storage providers (see the struct HSS_Storage definitions in `init/hss_boot_init.c`)

Each storage provider (QSPI, SPI, MMC, Payload, etc.) is registered in the `pStorages[]` array, and the boot process iterates through these to find a valid boot image.

You can add a custom storage provider by defining a new `struct HSS_Storage` with your own `getBootImage()` and `init()` functions. This is where you could implement logic to drive GPIOs for CS switching and perform checks for redundancy/fallback.

The boot code running in eNVM can be modified to support this, as the storage selection and boot image loading logic is extensible and modular.

* **ROM Boot Code Configuration : How can I ensure that the ROM boot code in eNVM only loads from a selected address range in the QSPI where I have placed different components (device tree, kernel image, root file system, and bootloader)? Where is this boot address/offset configured?**

The ROM boot code (e.g., HSS in eNVM) uses the storage provider's `getBootImage()` function to load the boot image.

The address/offset in QSPI from which the boot image is loaded is determined by the storage provider implementation and configuration macros generated from the `.config` file (e.g., `CONFIG_SERVICE_BOOT_QSPI_FLASH_OFFSET`).

You can control which address range is used by customizing these macros and the logic in your storage provider.

The boot image header is read from the configured offset, and the rest of the image is loaded accordingly. This is visible in functions like `getBootImageFromQSPI_()` and the use of `srcLBAOffset`. 

* **FPGA IO Control from eNVM : Once the FPGA fabric is configured and running, can the eNVM code drive/read the FPGA IO pins in a baremetal flow, similar to how the selection between eMMC and SD card is implemented in Icicle development kits?**

Once the FPGA fabric is configured, you can drive or read FPGA IO pins from HSS code, similar to how storage selection is implemented for eMMC/SD in the Icicle kit.

The HSS boot logic allows you to implement custom logic in your storage provider's `init()` or `getBootImage()` functions, including toggling GPIOs or interacting with the FPGA fabric as needed for storage selection or redundancy checks.

As an example, you could define two QSPI storage providers (main and redundant), each with its own CS logic and address range. The boot logic will try each in order, providing fallback if the first fails... This would look something like ...

```c
static struct HSS_Storage mainQSPIStorage = {
    .name = "MainQSPI",
    .getBootImage = getBootImageFromMainQSPI,
    .init = MainQSPIInit, // Set CS for main QSPI
    // ...
};

static struct HSS_Storage redundantQSPIStorage = {
    .name = "RedundantQSPI",    
    .getBootImage = getBootImageFromRedundantQSPI,    
    .init = RedundantQSPIInit, // Set CS for redundant QSPI    
    // ...
};

static struct HSS_Storage *pStorages[] = {
    &mainQSPIStorage,    
    &redundantQSPIStorage,    
    // ... other storages
};
```

The boot process will attempt to boot from `mainQSPIStorage` first, and if it fails, will automatically try `redundantQSPIStorage`.

Hopefully this helps!

griffini, 2025-10-28:
> Hello [@griffini](https://github.com/griffini)
> 
> Thanks for your detailed responses. While designing the full boot flow, I’ve encountered a few points I’d like to clarify:
> 
> * The HSS boot code resides in the eNVM, which I plan to modify to include redundancy and boot checks as we discussed. Since the eNVM size is limited to 128 KB, I’ll need to ensure the code stays within that limit, correct?

Correct.  The code in eNVM is mostly compressed to save space, with a tiny de-compressor at the start of it -- this is why it decompresses from eNVM to L2 Scratchpad to execute.

> * The HEX file generated by SoftConsole after modifying the HSS code can be linked in Libero, and the final .job file can then be programmed, this writes the HEX contents to the eNVM, right?

Correct.

> * The E51 core executes this HSS code from eNVM (after loading it into RAM), and then searches for the payload.bin at a specific offset. This payload.bin internally contains u-boot.bin,is that correct?

Correct... The E51 starts running the HSS decompressor code from eNVM, and decompresses the HSS to L2 Scratchpad. The HSS code itself runs from L2 Scratchpad. It searches the various configured boot sources (eMMC/SD, QSPI, SPI) for a payload.bin... The payload.bin contains boot images for one or more of the U54 harts.

> * I also noticed entries like `PolarFire-SoC-HSS::U-Boot` inside the payload. Is this just a naming convention used by the HSS or the actual HSS code that gets executed ( is this the code that takes care of loading things and different from HSS code in eNVM )  ? What other components can payload.bin include?

Just a naming convention, just to give a unique name... Could be anything...  payload.bin can also include ancilliary data blobs that are opaque to the HSS, but intended to be used by the payload for a particular U54. These blobs could be DTBs, for example.

> 
> ```
> root@icicle-kit-es:~# dd if=/dev/mmcblk0p2 of=/tmp/part2.bin bs=1M count=1
> 1+0 records in
> 1+0 records out
> 1048576 bytes (1.0 MB, 1.0 MiB) copied, 0.035792 s, 29.3 MB/s
> root@icicle-kit-es:~# hexdump -C /tmp/part2.bin | head -30
> 00000000  de c0 07 b0 01 00 00 00  c8 06 00 00 00 00 00 00  |................|
> 00000010  44 be 54 3b 00 00 00 00  60 06 00 00 00 00 00 00  |D.T;....`.......|
> 00000020  b0 06 00 00 00 00 00 00  00 00 20 80 00 00 00 00  |.......... .....|
> 00000030  01 00 00 00 00 00 00 00  01 00 00 00 00 00 00 00  |................|
> 00000040  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> 00000050  75 2d 62 6f 6f 74 2e 62  69 6e 00 00 00 00 00 00  |u-boot.bin......|
> 00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> *
> 00000150  00 00 20 80 00 00 00 00  01 00 00 00 00 00 00 00  |.. .............|
> 00000160  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> *
> 00000270  00 00 00 00 00 00 00 00  00 00 20 80 00 00 00 00  |.......... .....|
> 00000280  01 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> 00000290  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> *
> 000003a0  00 00 20 80 00 00 00 00  01 00 00 00 00 00 00 00  |.. .............|
> 000003b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> *
> 000004c0  00 00 00 00 00 00 00 00  50 6f 6c 61 72 46 69 72  |........PolarFir|
> 000004d0  65 2d 53 6f 43 2d 48 53  53 3a 3a 55 2d 42 6f 6f  |e-SoC-HSS::U-Boo|
> 000004e0  74 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |t...............|
> 000004f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> *
> 000005c0  00 00 00 00 00 00 00 00  08 68 0b 00 00 00 00 00  |.........h......|
> 000005d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> *
> 00000660  01 00 00 00 00 00 00 00  c8 06 00 00 00 00 00 00  |................|
> 00000670  00 00 20 80 00 00 00 00  3a 61 0b 00 00 00 00 00  |.. .....:a......|
> 00000680  99 5d 62 ba 00 00 00 00  00 00 00 00 00 00 00 00  |.]b.............|
> 00000690  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> root@icicle-kit-es:~# lsblk 
> NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
> mmcblk0      179:0    0  7.3G  0 disk 
> |-mmcblk0p1  179:1    0   64M  0 part 
> |-mmcblk0p2  179:2    0    8M  0 part 
> `-mmcblk0p3  179:3    0  7.2G  0 part /tmp/emmc0
> ```
> 
> * I plan to maintain a component-wise boot structure across QSPI partitions, as shown below. Does this layout look correct?
> 
> > Flash Layout (64MB):
> > ├─ 0x00000000 - 0x001FFFFF (8MB)   : payload ( contains only u-boot.bin ? )
> > ├─ 0x00200000 - 0x009FFFFF (8MB)   : kernel (raw Image + DTB)
> > └─ 0x00A00000 - 0x01FFFFFF (48MB)  : rootfs (UBI filesystem)
> > mtdparts=spi2.0:2m(payload),8m(kernel),22m(rootfs)

I can't really comment without knowing what you are booting. 8MB looks small for the kernel + DTB. And 48MiB looks very small indeed for a rootfs.

An 8 MB kernel partition can be enough, but it depends:
* A compressed kernel (zImage or Image.gz) with a small DTB can be 6–10 MB.
* If you include an initrd.img (built-in root filesystem), it can balloon to 20–30 MB easily.
* So 8 MB may be tight or insufficient for a modern kernel + DTB + optional initrd.

0x1FFFFF is not 8MiB? Shouldn't that be 0x7FFFFF?   The mtdparts seems to reflect the sizes, but your MB values seem wrong... Having said that, if your payload only contains u-boot.bin, you could probably trim this to 2MiB?

48 MB is quite small for a Linux rootfs, unlessyou’re using a very minimal build (BusyBox + small set of libs), and you’re storing everything compressed (UBI/UBIFS helps here).

griffini, 2025-11-10:
> Hello [@griffini](https://github.com/griffini) ,
> 
> Thanks for all the answers. Apologies for late response.
> 
> > 1.) I don't understand what you mean here? Lots of individual artefacts are built.
> 
> * The linux image ( fitimage + dtb ),rootfs and u-boot.env/.scr, are there makefiles to build these individually ?

Yes.  For Yocto questions, you're probably better off asking in https://github.com/linux4microchip/meta-mchp

But yes, you can do `bitbake <recipe>`. This will build the recipe and all its dependencies, performing all necessary tasks (fetch, configure, compile, package, etc.).

Other useful tasks:
* `bitbake -c configure <recipe>` - just configure the package.
* `bitbake -c compile <recipe>` - compile the package.
* `bitbake -c install <recipe>` - stage files into image directory (optional step before packaging)
* `bitbake -c package <recipe>` - creates .ipk/.rpm/.deb files as configured
* `bitbake -c clean <recipe>` -  clean previous builds.
* `bitbake -c rebuild <recipe>` - clean + compile.
* `bitbake -c fetch <recipe>` -  download source.

You can do:

````
bitbake -c clean u-boot-mchp
bitbake -c compile u-boot-mchp
bitbake u-boot-mchp
bitbake mpfs-linux
````
etc.

> 
> > 4.) apps (20 MB): if read‑only, I wonder if SquashFS might be more space‑efficient? UBIFS if writable?
> 
> * Didn't get this completely, can you explain more or give references ?

https://docs.kernel.org/filesystems/squashfs.html
https://docs.kernel.org/filesystems/ubifs.html

> 
> > 5.)  I'd suggest this is best done as a separate HSS service.
> 
> * How can this be done, is there any pre-defined process ?
> 

1. Look at services/ and copy an existing one - take `lockdown` or `ddr` as a base, and copy it to `services/<new_name>`.  
2. Edit `services/Kconfig`, `services/Makefile` to add it in to the build.
3. Edit `application/hart0/hss_register.c` to add it into `pGlobalStateMachines[]`. Just copy the existing patterns that are there. Now your state machine should run when you rebuild and reprogram the HSS.

ankrtlpixxel, 2025-11-17:
Yes,

Here it is.

MSS_GPIO_init(GPIO2_HI);

ankrtlpixxel, 2025-11-25:
Hello @griffini 

Further questions :

1.) Can we use any UART from the available list to get the HSS logs? 
```
void *HSS_UART_GetInstance(int hartid)
{
    mss_uart_instance_t *pUart;
    const bool postBoot = HSS_Trigger_IsNotified(EVENT_POST_BOOT);

    switch (hartid) {
    default:
        pUart = pUartDeviceList[0];
        break;

    case HSS_HART_E51:
        if (postBoot) {
            pUart = pUartDeviceList[CONFIG_UART_POST_BOOT];
        } else {
            pUart = pUartDeviceList[0];
        }
        break;

    case HSS_HART_U54_1:
        pUart = pUartDeviceList[1];
        break;

    case HSS_HART_U54_2:
        pUart = pUartDeviceList[2];
        break;

    case HSS_HART_U54_3:
        pUart = pUartDeviceList[3];
        break;

    case HSS_HART_U54_4:
        pUart = pUartDeviceList[4];
        break;
    }

    return pUart;
}
```
2.) For icicle kit-ES, I see CONFIG_UART_POST_BOOT=0. Does that mean it uses MMUART0 for all the HSS logs? 
3.) Since the Linux device tree uses serial1 as the stdout-path, can any other port be used for STDIO? Do we need to ensure these pins are routed as FPGA I/O from the MSS, or can they be driven directly by the MSS?
```
MMUART_0                                            FABRIC
MMUART_0_MODE                                       ASYNCHRONOUS
MMUART_0_MODEM                                      UNUSED
MMUART_0_OTHER                                      UNUSED
MMUART_1                                            FABRIC
MMUART_1_MODE                                       ASYNCHRONOUS
MMUART_1_MODEM                                      UNUSED
MMUART_1_OTHER                                      UNUSED
MMUART_2                                            FABRIC
MMUART_3                                            FABRIC
MMUART_4                                            FABRIC

```